iot-lab/iot-lab #176: Manage Robot profiles from cli-tools

cladmi, 2015-02-06:


cladmi, 2015-02-20:
Basic mobility profile that should be submitted

    {
        "consumption": {
            "average": 4,
            "current": false,
            "period": 8244,
            "power": false,
            "voltage": false
        },
        "mobility": {
            "site_name": "grenoble",
            "trajectory_name": "circuit1"
        },
        "nodearch": "m3",
        "power": "dc",
        "profilename": "gre_circuit1",
        "radio": null
    }

Full mobility profile that will be got after re-reading profile

    {
        "consumption": {
            "average": 4,
            "current": false,
            "period": 8244,
            "power": false,
            "voltage": false
        },
        "mobility": {
            "coordinates": [
                {
                    "name": "0",
                    "w": 0.0,
                    "x": 40.6,
                    "y": 34.2,
                    "z": 0.0
                },
                {
                    "name": "1",
                    "w": 3.14,
                    "x": 59.4,
                    "y": 34.2,
                    "z": 0.0
                }
            ],
            "points": [
                "0",
                "1"
            ],
            "site_name": "devgrenoble",
            "trajectory_name": "circuit1"
        },
        "nodearch": "m3",
        "power": "dc",
        "profilename": "gre_circuit1",
        "radio": null
    }

As we can get the full mobility from circuit list, sending full profile would be great to be symmetric, and allow testing.

fsaintma, 2015-04-20:
Get sites trajectory list  :  GET /robots/circuits
Remove authentication

cladmi, 2015-04-23:
Added in 1.6.0 branch https://github.com/iot-lab/cli-tools/tree/1.6.0

closed 2015-05-28 by fsaintma