iot-lab/iot-lab #180: Add support to start/stop the open node debugger on M3 nodes.

cladmi, 2015-02-16:


cladmi, 2015-02-20:
Release on Grenoble dev platform

fsaintma, 2015-04-21:
* Add tutorial page : https://www.iot-lab.info/?page_id=4377&preview=true
* You can reset/halt and flash firmware with gdb but it's impossible to add breakpoints.
* You must desactivate in openlab/platform/platform.c line 81 the sleep mode and recompile your firmware (https://groups.google.com/forum/#!topic/osdeve_mirror_rtos_ecos-discuss/_vmWTGdr1O8)
   // Enter SLEEP mode
  //asm volatile("wfi");
* Now when you debug tutorial_m3 firmware you have problem with event timer

closed 2015-05-28 by cladmi

cladmi, 2015-07-22:
You managed to add breakpoints in the tutorial but I couldn't do it, I got the following error:

```
(gdb) c
Continuing.
Warning:
Cannot insert hardware breakpoint 1.
Could not insert hardware breakpoints:
You may have requested too many hardware breakpoints/watchpoints.
```

I didn't deactivated the `asm volatile("wfi") however, maybe it comes from here.