iot-lab/iot-lab #183: Inconsistent "archi" between rest interfaces

cladmi, 2015-02-23:
In the rest interface, the "archi" parameter is sometimes contained in ('m3', 'a8', 'wsn430') and sometimes in ('m3:at86rf231', 'a8:at86rf231', 'wsn430:cc1101', 'wsn430:cc2420'). I will call them "simple" and "full".

+ Experiment submission:
    * logical -> full archi
    * physical -> simple archi
+ node-cli:
    * simple archi
+ Experiment info:
    * `--list` -> full archi
    * `--list-id` -> simple archi

When trying to script around this interfaces, it often requires to switch from the simple to the full archis and the opposite.

Maybe this can be improved.

cladmi, 2015-02-23:
The main problem is that `experiment info --list-id` is the only commands that easily returns nodes per site / archi but the archi is incomplete.

Currently, the solution to get infos on "sites -> archis_list" is to use '--list-id' or parse all the elements in '--list'.

cladmi, 2015-02-23:
Also, the 'site' is not directly available in 'experiment-cli info --list' as it's only in the network_address.

vandaele, 2019-06-17:
Should be ok now with our new API.
Transfer this issue to the testbed-manager project, to still check.