iot-lab/iot-lab #196: wsn430 -TinyOS 2.1.2

alesilva241, 2015-07-17:
I am trying install toolbox to work with tinyOS to wsn430, but I have installed in my PC tinyOS 2.1.2 and git tutorial doesn't work to me. Some one was in the same way? Can help me with this?  I tried work with tinyOS 2.1.1 from git tutorial but the repository link on git doesn't work too.

cladmi, 2015-07-22:
Hi Alexandra,
for a few weeks TinyOS removed the `http://hinrg.cs.jhu.edu/tinyos lenny main` repository and from what I saw the provided versions are not compatible with the TinyOS 2.1.1 version we are providing.
And so the support has been removed from the frontends.

According to an old mail on the mailing it may be easy to use them with the newer versions:
https://gist.github.com/cladmi/268a84e2998d34a22b4e

However I didn't had the time to really test it. 

If you validate it and make it work we would be interested to integrate your corrections.

Regards, Gaëtan

alesilva241, 2015-07-22:
Gaëtan, what is the indicated supported by wsn-430 iot-lab version of tinyOS?  TinyOS 2.1.2?

cladmi, 2015-07-24:
The supported version is the 2.1.1, however tinyOS does not support it anymore.
But as mentioned in the gist link, not so many changes should be required to work with 2.1.2.

lixinlu2000, 2015-08-24:
alesilva241, you can follow the link:http://tinyos.stanford.edu/tinyos-wiki/index.php/Installing_TinyOS_2.1.1#Step_4:_Install_TinyOS_toolchain
I have tried.

lixinlu2000, 2015-08-24:
Hi,cladmi, 
when I use: 
cd Apps_official_2/Blink 
make wsn430v13
on the ssh frontend, there is a error: error:make: *** No rule to make target 'wsn430v13'.
when I test Blink in my computer, I added the MAKERULES in my bashrc, it works, but in frontend, I did not find the makerules in wsn430_tinyos-2.1.1_port/support/make folder, so what can fix this error?
Regards
xinlu

cladmi, 2015-08-27:
Hi lixinlu2000,

the toolchain for tinyOS and the tinyOS source are not installed anymore on the frontend.
If you have a tutorial on how to install it that works, I can add it.

cladmi

alesilva241, 2015-12-09:
Hey guys I am back. I want so much test my applications in WSN430. Is there a way to run tinyOS 2.1.2 on this platform? Need I change something? Contribute with something? I wanna help! ;) Now...I have a lot of free time to coding.. :+1:

rodrigoct, 2018-09-17:
The supported version is the 2.1.1 yet?

schrein, 2019-06-17:
WSN430 are EOL nodes, we do not plan to work on TinyOS anymore. However, contribution are welcome. In the meantime, we close the issue.

closed 2019-06-17 by schrein