iot-lab/iot-lab #197: Write a script to allow running a 'serial_aggregator' in background for an experiment

cladmi, 2015-07-23:
Add a script that allows running a script during an experiment without having to keep the terminal running.

Idea on how to run this using screen and ssh:

```
SITE="grenoble"
EXP_ID="26016"
CMD="serial_aggregator -i ${EXP_ID}"
OUT_FILE="~/.iot-lab/${EXP_ID}/experiment_output.csv"

ssh ${SITE} "screen -dmS ${EXP_ID} bash -cx \"experiment-cli wait -i ${EXP_ID} && ${CMD} | tee -a ${OUT_FILE}\""
```

cladmi, 2015-07-23:
* Storing log for the whole bash script would help in case of early crash.
* Allow Stoping the experiment in case of crash/end of command

cladmi, 2015-08-24:
Ideas to add the feature on a testbed level:

Add a python script "exp_run_script.py [user] [exp_id|session_name] [script_path]"

* `su - user`
    * Should be runnable as the correct user directly, so no need to su.
* run a `screen` session named "exp_id|session_name"
* Pass experiment_id as an env variable
* run given script path
* Should allow running "serial_aggregator -i ${EXP_ID}"
    * PATH variable setting
    * arguments
    * Maybe run it inside a bash also to be able to evaluate "${EXP_ID}"
* It would lack the "experiment-cli wait" functionality as this would need to be managed by the testbed.

aabadie, 2019-06-17:
This is done, let's close this.

closed 2019-06-17 by aabadie