iot-lab/iot-lab #198: REST server breaking when testing `experiment-cli wait` error cases

cladmi, 2015-07-24:
I wanted to try to find the `experiment-cli wait` server error to handle them locally.
I launched ~20 experiment-cli wait to see the possible errors

After some time I get this error on some execution

```
$ experiment-cli wait
Waiting that experiment 26056 gets in state Running
HTTP Error 500: 
	Too many open files
```

And then on all other and every time I run a REST request I get:

```
 experiment-cli wait
HTTP Error 401: Unauthorized: Wrong login/password

	Register your login:password using `auth-cli`
```

[image]

It fixes itself after some time.

It looks like not correctly closed files gets garbage collected after a "long" time, or java taking it's time to really close the open files.

cladmi, 2015-07-24:
I previously already had issue after a long time use of only one 'experiment-cli wait' but it took longer.
By default it does requests with 5 seconds between each.

fsaintma, 2015-09-05:
Fix on dev platform and refactoring code. The bug is due to bad disconnect ldap connexions.

closed 2015-09-05 by fsaintma