iot-lab/iot-lab #205: serial_aggregator: --list gets ignored if used with --id

OlegHahm, 2016-01-23:
If the `serial_aggregator` gets called with `--id` or `-i` the list of nodes supplied by `--list/-l` is ignored.

cladmi, 2016-01-23:
That's strange, for me it gets applied correctly:

I give an experiment with two nodes with `--id` and a a third node using `--list`

```
harter@grenoble:~$ serial_aggregator -i 34911 -l grenoble,m3,80
1453551140.422809;Aggregator started
u
1453551142.008841;m3-7;Current node uid: 9883 (m3-7)
1453551142.009120;m3-7;
1453551142.009348;m3-80;Current node uid: 9476 (m3-80)
1453551142.009529;m3-80;
1453551142.009780;m3-9;Current node uid: 2162 (m3-9)
1453551142.009955;m3-9;

harter@grenoble:~$ experiment-cli get -ri -i 34911
{
    "items": [
        {
            "grenoble": {
                "m3": "7+9"
            }
        }
    ]
}
```

I tried on Grenoble, on which site did you had the problem.
Because it could really be an outdated version still running somewhere.

OlegHahm, 2016-01-23:
I'm currently working at the Lille site.

OlegHahm, 2016-01-23:
```
> serial_aggregator --id 34905 --list lille,m3,1-7+9-21
1453551867.243643;Aggregator started
1453551867.245047;m3-27;Connection closed
1453551867.245546;m3-27;Connection closed
m3-22;foo
1453551875.376944;m3-22;foo
1453551875.377946;m3-22;shell: command not found: foo
1453551876.099620;Stopping
.> serial_aggregator --list lille,m3,1-7+9-21
1453551882.653003;Aggregator started
m3-22;foo
1453551887.715831;Node not managed: m3-22
1453551890.999728;Stopping
```

OlegHahm, 2016-01-23:
Ah, now I see what you mean - maybe I just misinterpreted the documentation. With `--list` I can give an additional list of nodes, not specify an absolute set?

cladmi, 2016-01-23:
Yes, I'm adding `--id` nodes to `--list` nodes: https://github.com/iot-lab/aggregation-tools/blob/master/iotlabaggregator/common.py#L139

OlegHahm, 2016-01-23:
Maybe the online documentation could be a bit more verbose about this, otherwise I'm fine with closing this issue.

cladmi, 2016-01-23:
There is online documentation ? :o
You mean the tutorial or the README ?

OlegHahm, 2016-01-23:
I meant `serial_aggregator --help` ->
```
  -l NODES_LIST, --list NODES_LIST
                        nodes list
```

cladmi, 2016-01-23:
It used to be better before: https://www.iot-lab.info/tutorials/nodes-serial-link-aggregation/
I will take a look next week.

Suggestions on how to explain it are welcome :)

Thank you for the report.

cladmi, 2016-01-23:
However, for you initial problem, if you want to select all nodes but node m3-22, if you are not scared of connection error messages you can list nodes that aren't in your experiment, like `lille,m3,1-21+23-500`.

Also, as the redirection only allows one connection at a time and rejects the other ones, what I usually do is first run `nc` or `tunslip6` on one node, and then run `serial_aggregator` which would only be able connect to the other nodes and leave me with a non redirected node.

OlegHahm, 2016-01-23:
Thanks for the help and explanation. I will try to think about a better documentation. What does `serial_aggregator` do if I omit `--list` and `--id`? Aggregates all nodes at the current site?

OlegHahm, 2016-01-23:
And as an explanation for my initial problem: the `make iotlab-term` command in RIOT will always pass the `--id` parameter to serial_aggregator. I expected to get a subset by setting `IOTLAB_PHY_NODES` accordingly, but ended up with all nodes. However, I reserved 60 nodes, but wanted to use subsets of 20 nodes in a different manner. I should have probably just reserved three experiments with 20 nodes each, but I was too lazy. ;)

cladmi, 2016-01-23:
If you omit `--list` and `--id` it uses your current experiment nodes, as does `node-cli`.

I may also add a `--exclude` option as in `node-cli` to remove nodes from the list.

If I start changing the internal interface, I may re-add the option to specify multiple time `--id` to aggregate multiple experiments.

vandaele, 2019-06-17:
Still have some good ideas here.
I just transfer this issue to the `aggregation_tools` repository

vandaele, 2019-06-17:
Transfer this issue to `aggregation_tools` repository, still there's some good features ideas not yet implemented.