iot-lab/iot-lab #206: Provide possibility to trigger power measurement intervals via GPIO

authmillenon, 2016-03-01:
I'm not 100% sure if this is possible (or if this is the right way to start a feature request), but it would be really nice if one could trigger the start and end of a measurement interval for power consumption (for RSSI to probably) via the device's GPIO. The current system with time frames is helpful, but requires a lot of data analysis to map the consumption data to the actual application data.

cladmi, 2016-03-03:
Hi Martin, I think the best is to put this on the mailing list, I will answer there.

schrein, 2019-06-17:
We move this features request to the the iot-lab-gateway repository. It should be the starting point for developing this new feature.