iot-lab/iot-lab #212: Permission denied (publickey).

Alimihamadalqasim, 2016-12-25:
When I try to open ssh i get the following? please help?
I have made sure that I paste the public key as a single line , but does not work.
OpenSSH_6.6.1, OpenSSL 1.0.1f 6 Jan 2014
debug1: Reading configuration data /etc/ssh/ssh_config
debug1: /etc/ssh/ssh_config line 19: Applying options for *
debug1: Connecting to grenoble.iot-lab.info [194.199.16.167] port 22.
debug1: Connection established.
debug1: identity file /home/user/.ssh/id_rsa type 1
debug1: identity file /home/user/.ssh/id_rsa-cert type -1
debug1: identity file /home/user/.ssh/id_dsa type -1
debug1: identity file /home/user/.ssh/id_dsa-cert type -1
debug1: identity file /home/user/.ssh/id_ecdsa type -1
debug1: identity file /home/user/.ssh/id_ecdsa-cert type -1
debug1: identity file /home/user/.ssh/id_ed25519 type -1
debug1: identity file /home/user/.ssh/id_ed25519-cert type -1
debug1: Enabling compatibility mode for protocol 2.0
debug1: Local version string SSH-2.0-OpenSSH_6.6.1p1 Ubuntu-2ubuntu2.3
debug1: Remote protocol version 2.0, remote software version OpenSSH_6.7p1 Debian-5+deb8u3
debug1: match: OpenSSH_6.7p1 Debian-5+deb8u3 pat OpenSSH* compat 0x04000000
debug1: SSH2_MSG_KEXINIT sent
debug1: SSH2_MSG_KEXINIT received
debug1: kex: server->client aes128-ctr hmac-sha1-etm@openssh.com none
debug1: kex: client->server aes128-ctr hmac-sha1-etm@openssh.com none
debug1: sending SSH2_MSG_KEX_ECDH_INIT
debug1: expecting SSH2_MSG_KEX_ECDH_REPLY
debug1: Server host key: ECDSA b1:3f:0b:04:78:f6:72:e2:86:25:81:21:71:3f:59:39
The authenticity of host 'grenoble.iot-lab.info (194.199.16.167)' can't be established.
ECDSA key fingerprint is b1:3f:0b:04:78:f6:72:e2:86:25:81:21:71:3f:59:39.
Are you sure you want to continue connecting (yes/no)? yes
Warning: Permanently added 'grenoble.iot-lab.info,194.199.16.167' (ECDSA) to the list of known hosts.
debug1: ssh_ecdsa_verify: signature correct
debug1: SSH2_MSG_NEWKEYS sent
debug1: expecting SSH2_MSG_NEWKEYS
debug1: SSH2_MSG_NEWKEYS received
debug1: Roaming not allowed by server
debug1: SSH2_MSG_SERVICE_REQUEST sent
debug1: SSH2_MSG_SERVICE_ACCEPT received
debug1: Authentications that can continue: publickey
debug1: Next authentication method: publickey
debug1: Offering RSA public key: /home/user/.ssh/id_rsa
debug1: Authentications that can continue: publickey
debug1: Trying private key: /home/user/.ssh/id_dsa
debug1: Trying private key: /home/user/.ssh/id_ecdsa
debug1: Trying private key: /home/user/.ssh/id_ed25519
debug1: No more authentication methods to try.
Permission denied (publickey).

muhilanr, 2017-02-09:
I am encountering the same problem.  I am getting the same response when I try to ssh. Have you figured it out?

acpteixeira, 2017-02-09:
​Not yet (sadly)​

*Atenciosamente,*
*Best Regards,*

*Ana Cristina Pereira Teixeira*
+55 31 9 8559-4248
Skype User: ana-cris-bh

cladmi, 2017-02-09:
Please send your iot-lab username, email, ssh public key and the full ssh command and error message to admin mailing list: admin@iot-lab.info

Either there is no way for us to debug.

Regards,
IoT-LAB Team

muhilanr, 2017-02-09:
Hello,
  I finally figured that out.

  Sorry for spaming your mailbox. But I like to thank you for the reply. I
appreciate replying back to help the user.

anonyette, 2017-12-19:
Hi Muhilanr,

Can you please inform us how was your problem solved because i am facing the same exact problem!!

Thank you

closed 2019-06-17 by rienafairefr