iot-lab/iot-lab #213: vagrant image

rpnegro, 2017-01-05:
Hi, maybe I'm doing something wrong but it seen that vagrant doesn't find the image. 

Bringing machine 'default' up with 'virtualbox' provider...
==> default: Box 'iotlab/trusty64' could not be found. Attempting to find and in
stall...
    default: Box Provider: virtualbox
    default: Box Version: >= 0
==> default: Loading metadata for box 'iotlab/trusty64'
    default: URL: https://atlas.hashicorp.com/iotlab/trusty64
==> default: Adding box 'iotlab/trusty64' (v0.1) for provider: virtualbox
    default: Downloading: https://atlas.hashicorp.com/iotlab/boxes/trusty64/vers
ions/0.1/providers/virtualbox.box
    default: Progress: 0% (Rate: 0/s, Estimated time remaining: --:--:--):--)
An error occurred while downloading the remote file. The error
message, if any, is reproduced below. Please fix this error and try
again.

Could not resolve host: filesender.renater.frfichiers

Any suggestion?.

Thanks

Ricardo

fsaintma, 2017-01-06:
Hi
You are right, the box link was unavailable. I have just created a new one and it works again.
Best regards.

closed 2017-01-06 by fsaintma

rpnegro, 2017-01-10:
Thanks.

Regards,,

Ricardo Peláez Negro