iot-lab/iot-lab #214: Installation

vishal623, 2017-01-10:
Hello,

I am not able to install this on ubuntu. 

Can you provide manual.

I got this error.

vishal@vishal-OptiPlex-9020:~/Downloads/iot-lab-master$ make setup-contiki 
git clone https://github.com/iot-lab/contiki.git parts/contiki
Cloning into 'parts/contiki'...
remote: Counting objects: 101611, done.
error: RPC failed; curl 56 GnuTLS recv error (-9): A TLS packet with unexpected length was received.
fatal: The remote end hung up unexpectedly
fatal: early EOF
fatal: index-pack failed
Makefile:39: recipe for target 'parts/contiki' failed
make: *** [parts/contiki] Error 128

Thanks

cladmi, 2017-01-10:
Hello, maybe an issue with github for some time, or a broken connection.
In this setup, we are just doing some `git clone ...` as said in the output.

So I think you just need to retry the command.

vishal623, 2018-08-09:
I have already raised another issue regarding the same and it has been resolved.

closed 2018-08-09 by vishal623