iot-lab/iot-lab #215: Zigbee on M3 nodes

DyiDriss, 2017-06-12:
Hello i'm using M3 node and i'm wondering if any one have a simple code to send data and receive it on a Xbee connected on another laptop.
Thanks

aabadie, 2017-06-12:
Hi,

Can you elaborate a bit more, please ?
It would be great if you could provide more information about your setup. With the actual provided information, I fear that we won't be able to give you the right information to precisely answer what you are looking for.

What kind of M3 do you have ? How is the XBee connected to the Laptop ? Are they in the same room ? What protocol do you want to use to send your data ?

DyiDriss, 2017-06-12:
Hello,
I just started using M3, so i don't know very much about it, i run a program to mesure the temprature and the pression from LPS331AP implemented on the M3, 
i also know that there is a radio communication on it ( AT86RF231).
a part of my project is to send these mesures to a raspberry connected to Xbee S1.
so from what i understood that AT86xxx and the Xbee S1 both communicate on 802.15.4 network

aabadie, 2017-06-12:
I understand that you are using one of the M3 from (and inside) IoTLab, sorry if I misunderstood.

If your raspberry pi + XBee is not in the same room or out of the 802.15.4 radio range of the iotlab-M3, then you don't have to care about the XBee on the PI. You just need IPv6 and routing configuration to/from the PI and this depends on your local network.

I suggest that you read and follow the tutorial available for setting-up and IPv6 network with IoT-LAB:
- [Using RIOT on an A8-M3](https://www.iot-lab.info/tutorials/riot-public-ipv66lowpan-network-with-a8-m3-nodes/)
- [Using Contiki on M3](https://www.iot-lab.info/tutorials/basic-m3-nodes-contiki-uip-stack-with-public-ipv6-on-ssh-front-end/) or [on A8-M3](https://www.iot-lab.info/tutorials/a8-public-ipv6/)

DyiDriss, 2017-06-12:
Ok so from what i read there is to way to do experiments on M3.
i'm developping directly on my laptop on linux. so i think that these tutos do not work in my case.
about the range:  the Xbee is just next to my M3 ( same room).
PS: I'm not familiar with networks i'm just starting learning it by my own
Thank you a lot

aabadie, 2017-06-12:
Ok, but how can you have an M3 ? (I guess you work with someone from IoTLAB ?)

What kind of Xbee do you have ? on USB ?

What you need is to  configure your XBee as a border router from 802.15.4 wireless network to usual ethernet network. I don't know how to do that with an Xbee. I know it's possible with an M3 using RIOT. See [this example](https://github.com/RIOT-OS/RIOT/tree/master/examples/gnrc_border_router). It should be also possible with Contiki but this is out of my knowledge.

There was someone interested in using XBee on USB as a border router in the Linux community. I don't know if we could get it to work. Here is the PR he was working on: https://github.com/RIOT-OS/RIOT/pull/4448

DyiDriss, 2017-06-12:
i have a Xbee S1 i did [ https://michael.bouvy.net/blog/en/2013/04/02/raspberry-pi-xbee-uart-serial-howto/ ](this) configuration.
i'm going to study the example that you provided me, i'll let you know if i advanced in my work
Thanks

aabadie, 2019-06-17:
From my understanding, this is not an iot-lab issue. I'm closing this.

closed 2019-06-17 by aabadie