iot-lab/iot-lab #217: robot-cli HTTP Error 500

schrein, 2017-11-13:
Link between robot-cli and REST server seems to be broken. 

I can submit an experiment with circuit, that's ok robot is moving:

```
$ experiment-cli submit -d 20 -n test_robot -l strasbourg,m3,97,mobility=v_line_6
{
    "id": 86583
}

$ experiment-cli get -ri
{
    "items": [
        {
            "strasbourg": {
                "m3": "97"
            }
        }
    ]
}
```
But I can not interact with robot (status, update): 

```
$ robot-cli status -l strasbourg,m3,97
HTTP Error 500:
```

```
$ robot-cli update -l strasbourg,m3,97 strasbourg,v_line_6
HTTP Error 500:
	Your update mobility command is not valid
```

fsaintma, 2017-11-13:
Hi

I test on the DEV platform and for robot status command we have a regression bug. I will fix it and deploy testbed-manager fix.

For the second update mobility command (robot-cli update strasbourg,v_line_6) it's just because you switch the site name and mobility name parameters.

You have launched robot-cli update SITE,NAME and if you view below the command-line help it's NAME,SITE and the testbed-manager didn't find any site name with v_line_6.
 
$ robot-cli update -h
Using custom api_url: https://devwww.iot-lab.info/rest/
usage: robot-cli update [-h] [-e EXCLUDE_NODES_LIST | -l NODES_LIST]
                        [-i EXPERIMENT_ID]
                        NAME,SITE

schrein, 2017-11-13:
> Le 13 nov. 2017 à 17:26, Saint-Marcel <notifications@github.com> a écrit :
> 
> Hi
> 
> I test on the DEV platform and for robot status command we have a regression bug. I will fix it and deploy testbed-manager fix.
> 

Thanks !

> For the second update mobility command (robot-cli update strasbourg,v_line_6) it's just because you switch the site name and mobility name parameters.
>
> You have launched robot-cli update SITE,NAME [...]
Ok sorry I missed that ! I just follow the tutorial which is currently wrong...

https://www.iot-lab.info/tutorials/robot-cli-client/#Update_mobiliy <https://www.iot-lab.info/tutorials/robot-cli-client/#Update_mobiliy>

schrein, 2017-11-13:
FYI the bug still present with valid syntax

```
schreine@strasbourg:~$ robot-cli update -l strasbourg,m3,97 v_line_6,strasbourg
HTTP Error 500:
```

fsaintma, 2017-11-13:
Really strange I test it with success on the DEV platform. So I will try to understand what is the problem 
tomorrow.

schrein, 2018-09-20:
it's working again.

closed 2018-09-20 by schrein