iot-lab/iot-lab #220: How to Upload my code in FIT/IoT-LAB

vishal623, 2018-05-16:
Hello sir,

I want to test my algorithm in your environment. But I am unable to upload it. In my system, I have Z1 mote and in your premises m3 or wsn430. I want to use m3-cortex. 

How can I compile my code to m3-cortex, so it is available for testing? Otherwise, I have to upload my code to your Contiki repo. 

Please help me regarding this. 

Thanks
Vishal

rienafairefr, 2018-05-16:
Hi Vishal,

Yes, we have different types of boards available on the testbed, see here https://www.iot-lab.info/hardware/#iot-lab-node, most based on m3 or wsn430

To compile a Contiki-based application for m3 nodes and upload it and run it on the testbed you can follow the tutorial here https://www.iot-lab.info/tutorials/contiki-compilation/ 

To use your own code, you just create a project folder and have the makefile point to the correct Contiki root, in the Makefile   CONTIKI=where_you_cloned_iot-lab_contiki    That way you don't need to upload to iot-lab/contiki

Best,

Matthieu
FIT IoT-Lab team

vishal623, 2018-05-16:
I have done the steps as you mentioned. In project folder, I edited the makefile and pointed Contiki to where I cloned from iot-lab. 

Now, I want to access it. How can I? 
Are there any steps for the same?

When I compile Contiki for m3 nodes, it always shows me old Contiki, not Contiki-3.0.

rienafairefr, 2018-05-16:
It's explicited in the tutorial: 
- compile your application (make TARGET=iotlab-m3 in your project dir)
- submit an experiment on iot-lab with the elf (experiment-submit [...])
- get results about your experiment :
https://www.iot-lab.info/tutorials/ sections nodes monitoring or serial-link-aggregation
You can get the serial link for a single node using `nc m3-xxx 20000` when the experiment is started

You can go through all the tutorials to get familiarised with the platform and testbed if you need to.

The Contiki version that is supported on M3 nodes on iot-lab is only version 2.6 for now. We might move and support contiki-ng directly and skip Contiki 3.0, does your code absolutely need 3.0 or can it run on 2.6 ?

vishal623, 2018-05-16:
Now it is working. I have removed the old iot-lab and clone it again. Now, there is only one folder like er-coap in apps directory. 

 "you just create a project folder and have the makefile point to the correct Contiki root, in the Makefile CONTIKI=where_you_cloned_iot-lab_contiki " for this I have created folder in my systems contiki and in makefile I wrote the path of iot-lab contiki.

Now, how can I access it from genoble site?

rienafairefr, 2018-05-16:
the project folder can be anywhere, doesnt have to be inside any Contiki hierarchy. You probably shouldnt mix your system contiki with iot-lab's.
What do you mean by access it ?

vishal623, 2018-05-16:
ok. I already mix it.

rienafairefr, 2018-05-16:
normally you should just need the openlab repo next to the contiki repo. Using github.com/iot-lab/iot-lab and make setup-contiki as explained in the tutorial should make this for you.
you shouldnt have to modify contiki/platform/openlab/Makefile.include or tinker with symlink 

```
git clone git@github.com:iot-lab/iot-lab.git
cd iot-lab
make setup-contiki
```
then in your makefile, in your project, CONTIKI=path_to_iotlab_contiki

rienafairefr, 2018-05-16:
OK it seems I was mistaken, iot-lab/contiki master branch normally is based on Contiki 3.0, so I'm not sure why you're getting old contiki

vishal623, 2018-08-07:
Thanks, sir,

I am able to compile the code. I have to close the issue.

closed 2018-08-07 by vishal623