iot-lab/iot-lab #221: How to save my data

MuyuXiaoxiang, 2018-06-01:
Hello sir,

I want to test my algorithm in your environment. But I do not know how to save the data in my computer, I have the wsn430 mote to test my experiment.

Please help me regarding this.

Thanks
zhefan

rienafairefr, 2018-06-01:
Hi Zefan,

Can you precise what kind of data you're after ?
For WSN430 motes, I'm afraid the only data you can fetch is serial link output.
Have you completed all the relevant tutorials ?
https://www.iot-lab.info/tutorials/
You can look at the Serial Aggregator tutorial, then maybe the Run Script tutorial il you want to set up unattended serial link data collection. Don't forget their prerequisites as well.
If you want to gather data about consumption and radio packets, you should use other motes, the wsn430 are being phased out after a decade of service.

MuyuXiaoxiang, 2018-06-02:
OK！Thank you,sir!

MuyuXiaoxiang, 2018-06-02:
Thank you，Sir!
 I want to store the temperature data on my computer.I will try the method you provided.Thank you.

closed 2018-06-05 by rienafairefr