iot-lab/iot-lab #226: A8 nodes deployment failure

druzmieres, 2018-06-26:
I've been trying to run an experiment using A8 nodes but the deployment is failing every time. I tried having a successful deployment with and without an association and it fails in both cases. It fails in the old web portal and in the new one too. M3 nodes seem to be working correctly.

fsaintma, 2018-06-27:
Hi
Indeed I have just saw the testbed logs and it seems that we encountered some network issues. I launch an experiment this morning and it works normally. Should you retry again ?
Sorry for the inconvenience.
Best regards.

druzmieres, 2018-07-03:
Hi,
Are you having the same networks issues again? I'm having the same problem now.
Regards.

fsaintma, 2018-07-04:
Hi @druzmieres , I have just investigated again and I found that the problem is only for your account. So the last time I made a mistake and the problem seems to be due to your home directory on the Grenoble site. Indeed the A8 directory was removed (by you ?) and it was mandatory for A8 experiment nodes setup. So I have just created it again with good permissions and I hope that it will fix your problem.

druzmieres, 2018-07-07:
@fsaintma Thank you, I'm new at this and probably deleted that folder when I was trying to solve some other issue. I didn't know the folder was mandatory. Actually, I thought it was automatically generated.

Now I can finally run some experiments. :)

aabadie, 2019-06-17:
Seems that this is fixed. Closing.

closed 2019-06-17 by aabadie