iot-lab/iot-lab #229: Unable to Login on FIT/IoT-LAB

vishal623, 2018-08-28:
Hello,

I am unable to access FIT/IoT-LAB with either of the following accounts.

1)
username: rathod
email: rathodvishal78@gmail.com
ssh public key:
ssh-rsa
AAAAB3NzaC1yc2EAAAADAQABAAABAQDO5QzsL0MNNb3yrOgpZIBJYf/O2UUbR3T22kG8XxAEZr0dBFKhOceQxor2zDVmxsG7ounF0uvT7XZn9s0fz0VA0T0yEdhYQLFnywRx2bKagAuWYF9rvzUyV+5i7R+s12XWJ3je6MBiPUG93TFzwIhmsz9azbHu36FQs94Zu7MtFlzePMmaNXSIyOkLIxH753UK8GsFWojPI9fgyIAf+U4fqqrH/3J4gv0JbozLEMDjgiS+2p9y/WfnIvM0bMgPVc5h9Gu8od6QHzvgK5eSPG+86dylYSRHPg5tR6CRvr/vwGEDo4XYH6IlH8gtdO4QOgKmrE/1kGVanGl5xRTFfdJt
vishal@vishal-Z51-70
ssh command: ssh rathod@grenoble.iot-lab.info

2)
username: vishal
email: vishalrathod.cs15f11@nitk.edu.in
ssh public key:
ssh-rsa
AAAAB3NzaC1yc2EAAAADAQABAAABAQDO5QzsL0MNNb3yrOgpZIBJYf/O2UUbR3T22kG8XxAEZr0dBFKhOceQxor2zDVmxsG7ounF0uvT7XZn9s0fz0VA0T0yEdhYQLFnywRx2bKagAuWYF9rvzUyV+5i7R+s12XWJ3je6MBiPUG93TFzwIhmsz9azbHu36FQs94Zu7MtFlzePMmaNXSIyOkLIxH753UK8GsFWojPI9fgyIAf+U4fqqrH/3J4gv0JbozLEMDjgiS+2p9y/WfnIvM0bMgPVc5h9Gu8od6QHzvgK5eSPG+86dylYSRHPg5tR6CRvr/vwGEDo4XYH6IlH8gtdO4QOgKmrE/1kGVanGl5xRTFfdJt
vishal@vishal-Z51-70
ssh command: ssh vishal@paris.iot-lab.info

When I tried to log in with either of these accounts it says wrong login or password. When I tried to reset the password, it shows me error that invalid email ID. I was facing this problem today. Before that, I was not able to access the lab from my Institute network but I can access the lab outside of the Institute. 

Is it possible that my name has been in the blacklist or they blocked me for accessing the lab?

temur-zhao, 2018-09-06:
hello, I also met this issue today, it's almost one week, I'm not sure do you solve this issue or not, my solution is use command line 'iotlab-auth -u USERNAME' to change the password, actually I think 'change' is not correct, 'register' is better, hope it will be helphul.

vishal623, 2018-09-06:
Hello sir,

Thanks for the reply.

I have resolved this issue. I have generated the new password for my
username and run the iotlab-auth -u USERNAME command and it works well for
me.

vishal623, 2018-09-07:
If possible, then I want some help.  

I want to know, How can I run my algorithm in iot-lab? I have tried, but I am not able to do it.

temur-zhao, 2018-09-07:
sorry I'm pretty new to this project, what's kind of algorithm do you want to ues? is it means you want to organize some nodes by hand?

vishal623, 2018-09-08:
Thanks.

I want to upload my code which is based on CoAP on some nodes and try to
execute. For that, I already cloned the iot-lab repo and compiled it. But I
am unable to do it.

temur-zhao, 2018-09-09:
Hi, I am also trying to slove this problem for now, I will let you now if I have a way to do it.

vishal623, 2018-09-10:
@temur-zhao 
This is the login related query, and that has been resolved, so I close this issue and create a new one with the error that I was facing.

closed 2018-09-10 by vishal623