iot-lab/iot-lab #230: MAC sequence number error in TSCH (M3 node)

seungbumz, 2018-09-04:
I am testing TSCH (Specifically Orchestra from SenSys 2015) in IoT-LAB using M3 nodes. 

However, I found out continuous duplication of MAC sequence number in Rx-side.

To figure out what the problem is, I firstly checked seq. numbers in MAC frame in Tx-side, just before "NETSTACK_RADIO.transmit". And I found it works well.  

Then, I looked into "NETSTACK_RADIO.read" in Rx-side. 

The problem was, MAC seq. brought from Rx FIFO is always a constant value, "129". 
(However, the other fields in the frame is delivered perfectly.)

It is never changed even when seq. number in Tx-side increases. 

Since the code works well in Cooja simulator, I thought it is the problem of the radio driver of M3 node.
.
So I also investigated "radio-rf2xx.c" closely. But I could not find any reason for this problem.

Can you help me?

simonduq, 2018-09-04:
Never seen this before.

So you're saying that if you like at a byte dump of the packet right before Tx vs. right after Rx, they are identical except for the seqno byte? Only thing that comes to mind would be a driver bug, missing flush, or mis-configuration or something. Could you show here full packet dumps, Tx and Rx side, for several consecutive packets?

seungbumz, 2018-09-05:
In my experiment, 
Node 1 (#1) is root, and Node 2 (#2) is a child

Firstly, #1 transmits EB (Enhanced beacon), and #2 receives it and is associated to #1.

Then, KA (Keep-alive) packet is sent by #2, towards #1.

For the 1st KA packet, 
packet dump of Tx (#2): 21 ec 01 cd ab 01 00 00 00 00 00 00 02 02 00 00 00 00 00 00 02
packet dump of Rx (#1): 21 ec 81 cd ab 01 00 00 00 00 00 00 02 02 00 00 00 00 00 00 02

After re-transmissions, the packet is lost, since #1 sends back ACK with seq: 0x81 (=129) and #2 ignores this ACK due to different seq with its own seq (1).  

For the 2nd KA packet,
packet dump of Tx (#2): 21 ec 02 cd ab 01 00 00 00 00 00 00 02 02 00 00 00 00 00 00 02
packet dump of Rx (#1): 21 ec 81 cd ab 01 00 00 00 00 00 00 02 02 00 00 00 00 00 00 02

For the DIO broadcast (by #1),
packet dump of Tx (#1): 01 e9 cd ab ff ff cd ab 01 00 00 00 00 00 00 02 7a 3b 3a 1a 9b ...
packet dump of Rx (#2): 01 e9 81 ab ff ff cd ab 01 00 00 00 00 00 00 02 7a 3b 3a 1a 9b ...
 
What is the problem for seq number only?

seungbumz, 2018-09-06:
I got it, I should have set RF2XX_SOFT_PREPARE to 0  for M3 nodes.

closed 2018-09-06 by seungbumz

simonduq, 2018-09-06:
Wait, can you elaborate a bit?
Is there a patch needed to fix the bug?

seungbumz, 2018-09-06:
When I looked into project-conf.h in contiki directory from iot-lab (ex. https://github.com/iot-lab/contiki/blob/master/examples/ipv6/rpl-tsch/project-conf.h), I found that 

/* Needed for IoT-LAB M3 nodes */
#undef RF2XX_SOFT_PREPARE
#define RF2XX_SOFT_PREPARE 0
#undef RF2XX_WITH_TSCH
#define RF2XX_WITH_TSCH 1

I used to implement using common contiki directory, and there wasn't any comment like this.

The error i mentioned occurs when RF2XX_SOFT_PREPARE is set to 1 (it is default setting).

simonduq, 2018-09-06:
ah ok.