iot-lab/iot-lab #235: Unable to access a node using ssh

npcode15, 2018-09-12:
Hello, I am trying to follow the steps from the Tutorial, "Debugging with M3 Node" (https://www.iot-lab.info/tutorials/debugger-on-m3-nodes/)

I am stuck at the step when I need to access me-node using nc i.e. the command:
nc m3-<id> 20000.

This command does nothing. Please help. Thanks

aabadie, 2018-09-12:
Hi @npcode15,

Thanks for using IoT-LAB.

The command `nc m3-<id> 20000` is just an example. You must replace `<id>` by the id of the node that is booked for your experiment. In your case, it was 100 or 101.

So the command should be either `nc m3-100 20000` or `nc m3-101 20000`. Once the connection is opened, you can type `h` to get the list of available commands on the firmware that runs on the M3 node.

rienafairefr, 2018-09-14:
Hi @npcode15,
Do you remember which experiment this was happening with ? Looking at your deployment status
(by yourself you can see it too: in your dashboard, you can click on the experiment Id and get a page at https://www.iot-lab.info/testbed-beta/#/experiments/[experiment_id]), I see: 
* Some experiments failed to deploy ("Error") (e.g. 134042)
* Some unknown error ocurred (code "Unknown", which we will investigate) (e.g. 133509)
* Some experiments don't have firmwares (133236), in that case the firmware on the node is 'idle', which wouldn't print anything on the serial link.

If the firmware on your node doesn't print anything to the serial link (disabling printing on serial is an option that can be enabled in some OSs to save power depending on how you build), you will not see anything either.

npcode15, 2018-09-17:
Hey @rienafairefr,

Submitted another experiment: 134333 (would end before you could check probably).
Same issue as before. Uploaded a firmware, nodes deployed successfully yet unable to access the nodes with nc command.

Verified with native deployment, It prints content on the serial node. So, technically should work.

Thanks.

rienafairefr, 2018-09-17:
Are you sure of the id 134333 ? You're user omeliyan ?

rienafairefr, 2018-09-17:
nevermind, found your experiment. There seems to be a problem with the control node on that gateway unfortunately, so it's probable your node is not even booted. We'll look into it. Can you try with other nodes instead of m3-100 on grenoble ?

npcode15, 2018-09-17:
@rienafairefr,

Yes, the id was 134333,  User: npandey
[image].

Submitted again: Exp Id: 134338.
[image]

Nothing happens.

ichatz, 2018-11-28:
Hi all, I am getting exactly the same problem. I tried with nodes in Grenoble (e.g., experiment #143892) and also in Saclay (e.g., experiment #143896).

aabadie, 2019-06-17:
Your problem was fixed in the testbed recently. Closing this.

closed 2019-06-17 by aabadie