iot-lab/iot-lab #236: Energest (energy estimate) module on iotlab-m3 platform

hieu1204, 2018-09-13:
I'm trying to measure the amount of time consumed by CPU and Radio (i.e. LPM, Active) on the iotlab-m3 platform but I stuck by some problems
1. My code worked well on Z1 mote by implementing core/sys/energest.h 

``
static` uint32_t last_tx, last_rx, last_time;
static uint32_t delta_tx, delta_rx, delta_time;
static uint32_t curr_tx, curr_rx, curr_time;

/*---------------------------------------------------------------------------*/
void
simple_energest_init()
{
    energest_flush();
    last_tx = energest_type_time(ENERGEST_TYPE_TRANSMIT);
    last_rx = energest_type_time(ENERGEST_TYPE_LISTEN);
    last_time = energest_type_time(ENERGEST_TYPE_CPU) + energest_type_time(ENERGEST_TYPE_LPM);
}
/*---------------------------------------------------------------------------*/
void
simple_energest_step(int verbose)
{
    static uint16_t cnt;
    energest_flush();

    curr_tx = energest_type_time(ENERGEST_TYPE_TRANSMIT);
    curr_rx = energest_type_time(ENERGEST_TYPE_LISTEN);
    curr_time = energest_type_time(ENERGEST_TYPE_CPU) + energest_type_time(ENERGEST_TYPE_LPM);

    delta_tx = curr_tx - last_tx;
    delta_rx = curr_rx - last_rx;
    delta_time = curr_time - last_time;

    last_tx = curr_tx;
    last_rx = curr_rx;
    last_time = curr_time;

    if(verbose) {
        uint32_t fraction = (1000ul * (delta_tx + delta_rx)) / delta_time;
        LOG("Duty Cycle: [%u %u] %8lu +%8lu /%8lu (%lu permil)\n",
            node_id,
            cnt++,
            delta_tx, delta_rx, delta_time,
            fraction
        );
    }
}
``

Log output on Z1 node
`
01:06.490	ID:10	Duty Cycle: [10 1]      798 +  793328 / 1966633 (403 permil)
`
Log output on M3 node
``
Duty Cycle: [3 1]        0 +       0 /       0 (0 permil)
``
2. I also tested the powertrace app but the outputs are all zero too.

3. I found that the problem may be caused by the differences within the loop in the main files of two platform, i.g. in M3 ( platform/openlab/openlab-main.c )
`
while(1)
    {
        int r;
        do
        {
            watchdog_periodic();
            r = process_run();
        } while(r > 0);
        idle_count++;
    }
`
in Z1 ( platform/z1/contiki-z1-main.c) 
`
 while(1) {
    int r;
    do {
      /* Reset watchdog. */
      watchdog_periodic();
      r = process_run();
    } while(r > 0);

    /*
     * Idle processing.
     */
    int s = splhigh();    /* Disable interrupts. */
    /* uart0_active is for avoiding LPM3 when still sending or receiving */
    if(process_nevents() != 0 || uart0_active()) {
      splx(s);      /* Re-enable interrupts. */
    } else {
      static unsigned long irq_energest = 0;

#if DCOSYNCH_CONF_ENABLED
      /* before going down to sleep possibly do some management */
      if(timer_expired(&mgt_timer)) {
        timer_reset(&mgt_timer);
        msp430_sync_dco();
      }
#endif

      /* Re-enable interrupts and go to sleep atomically. */
      ENERGEST_SWITCH(ENERGEST_TYPE_CPU, ENERGEST_TYPE_LPM);
      /* We only want to measure the processing done in IRQs when we
         are asleep, so we discard the processing time done when we
         were awake. */
      energest_type_set(ENERGEST_TYPE_IRQ, irq_energest);
      watchdog_stop();
      _BIS_SR(GIE | SCG0 | SCG1 | CPUOFF); /* LPM3 sleep. This
                                              statement will block
                                              until the CPU is
                                              woken up by an
                                              interrupt that sets
                                              the wake up flag. */

      /* We get the current processing time for interrupts that was
         done during the LPM and store it for next time around.  */
      dint(); /*disable interrupt*/
      irq_energest = energest_type_time(ENERGEST_TYPE_IRQ);
      eint(); /*enable interrput */
      watchdog_start();
      ENERGEST_SWITCH(ENERGEST_TYPE_LPM, ENERGEST_TYPE_CPU);
    }
  }
`
I have no idea of some functions which are _BIS_SR(), dint() and eint(). 
Is there someone has overcome this kind of problems?

fsaintma, 2018-09-13:
Hi I suggest you to post your question at users@iot-lab.info mailing-list.

rienafairefr, 2018-09-13:
Hi,

seconding @fsaintma , this is more of an issue with the Contiki-OpenLab codebase, the implementation of the iotlab M3 platform didn't implement or try to implement any support for energest. 

You have the benefit of real hardware deployed with real instrumentation, I recommend you use the power monitoring feature of the control node if that suits your purpose.
We at FUN team published a paper recently [here](https://doi.org/10.3390/s18082472) that roughly describes the setup. The source code of the power monitoring aggregator is [here](https://github.com/rienafairefr/jad_fit), you might be able to fit it for your purpose if you wish.
That would give you the overall power consumption of your code running on the testbed. Of course, if you need to know the ratio between cpu and radio, this would not be suitable.

hieu1204, 2018-09-13:
@fsaintma thanks for your advice but I also want to study the radio duty cycling, so I think it would be better if there is a way to implement energest module.

rienafairefr, 2018-09-13:
There is a way, but you'd need to dig into the openlab repo, and add the appropriate energest configurations and hooks in the appropriate places.
An option might be using the Firefly boards, there is a dozen in Lille, or other boards, the support for energest in contiki/contiki-ng might be there already.

hieu1204, 2018-09-14:
Managed it by adding some energest measurement functions on CPU main and radio driver. I've been testing and correcting it. Thanks for your help.

closed 2019-06-17 by schrein