iot-lab/iot-lab #237: Date issue with mobile node experiment

Ymonse, 2018-09-13:
Hello,

I have a problem when I try to launch an experiment on a mobile.

I used that command 
iotlab-experiment submit -d 20 -n simple_robot -l 1,archi=m3:at86rf231+mobile=1+site=grenoble,simple_robot.elf,mobility=square1
But the date on the dashboard is : 01/01/1970, so the experiment can’t be launched.

Thanks for the help.

Best regards,
Ymonse

rienafairefr, 2018-09-14:
AFAIK, there are no more mobile nodes in the Grenoble site, so the experiment was probably submitted but can't ever start, and the 01/01/1970 (unix timestamp 0) reservation might show up, this is a bug we should show a failure of some sort, not a 01/01/1970 reservation. There are no circuit named square1 on Grenoble either, so that shouldn't work either.

You can use the mobile nodes on Lille website, and don't hesitate to contact us if there are any issues.

I can't seem to correlate you to your user on IoT-LAB. 
What's your username and the experiment number that shows that bug ? 

Matthieu
FIT IoT-LAB team

Ymonse, 2018-09-17:
Hello, 

Thanks for the response.

In fact the mobility argument  was circuit1 and not square1 for the grenoble site.

I tried to launch a mobile node on lille with the simple_robot firmware, although the status of the experiment is "running" the deployment's status is "failure", is it a correct behavior ?

The user and experiment id are younsi, 134074 for the precedent test and 134343 for the new one.

Ymonse

rienafairefr, 2018-09-17:
Hi,
try on the other robots (258 or 259), I think the 257 has a problem, I'm on it.
You can't really deploy a ROS application. What you can do is control the robots through pre-defined or user customised circuits. For now the creation or modification of user-defined circuits is still being deployed (you can do it by calling the REST api, but cli-tools are not yet updated). 
You'll soon be able to create custom models for your robots, having it do something a bit more complicated than just follow a circuit. But for now, full ROS is not on the roadmap, we have our own ROS application to drive the robots.

Matthieu

Ymonse, 2018-09-17:
Hi,
Thank you for the help.
I tried with the 258 node and it's working.
Thank you for the clarification about ROS.

Ymonse

closed 2018-09-17 by Ymonse