iot-lab/iot-lab #239: Read from Serial Port

npcode15, 2018-09-18:
Hello, 

**Setup:** 2 M3 nodes, Grenoble site

what is the best way to read data from serial ports?
Assuming, I have 2 nodes: m3-100, m3-101, their port ids would be dev/tty0 and dev/tty1 respectively?

Please help.

rienafairefr, 2018-09-18:
when on the SSH frontend, /dev/tty* don't redirect to any nodes. They are not physically connected.

As you've seen in other tutorials, the serial data is redirected on the TCP port 20000 for each node (you can use `nc m3-100 20000` to get to the serial link of node m3-100)

You could manually open multiple links to the nodes through e.g. socat and shell pipes, but we've built a serial aggregator to simplify this. Tutorial is [here](https://www.iot-lab.info/tutorials/nodes-serial-link-aggregation)

`serial_aggregator > data.log` will dump the data. You can also write some Python code to extend the aggregator tools (code is open source [here](https://github.com/iot-lab/aggregation-tools) so that you fetch only the data you want and save that to a CSV, if that's what you want. But usually what we do is let the serial aggregator run, piped to a log file, `serial_aggregator > logs.log` then parse the log file to extract only what we need.

npcode15, 2018-09-19:
@rienafairefr 
Thanks.

closed 2018-09-19 by npcode15