iot-lab/iot-lab #240: Connection Refused

npcode15, 2018-09-18:
Exp ID: 134594, 134450

[image]

Basically, Exp 134594 is a reload from 134409 which was working just fine. I was able to access nodes without any issue.

npcode15, 2018-09-25:
This is issue is still troubling me.
[image]

Experiment IDs - 135198 and 135200

Please help ASAP. Thanks

rienafairefr, 2018-09-25:
Sometimes you get this error because two processes try to connect to a node at the same time, is that something you are doing ? You can't have two serial aggregators, or a serial aggregator+ a netcat.

npcode15, 2018-09-25:
@rienafairefr Yes, I had a netcat and aggregator opened at this instance. But, it does deny connections without having multiple connections opened as well. But, thanks.

schrein, 2019-06-17:
We identify an issue with this problem. It should be fix now.

closed 2019-06-17 by schrein

rienafairefr, 2019-06-17:
Hi @npcode15, there was an issue on the testbed due to SSH servers getting too many connections, it should be fixed now, don't hesitate to re-open a new one if you still get an error