iot-lab/iot-lab #242: Missing features | New Experiment Portal

npcode15, 2018-09-28:
Hello, IoTLab Testbed was updated and I am not able to find some of the features present earlier in the portal.

**Example:**
1. Firmware cannot be updated after submitting the experiment.
2. Nodes cannot be reset when the experiment is running.

These are necessary tasks for me. I need to know the IP addresses of the nodes first so that I can put them in my code before compiling and uploading. In this scenario, updating firmware after submitting the experiment came in handy as I was able to submit a dummy firmware file to find the IP address.

If these things have shifted somewhere please let me know. Thank you.

fsaintma, 2018-09-28:
Hi
When your experiment is running and you click on it in your dashboard list you should see experiment details with nodes and a button at the right "Actions on selected nodes". When you have selected some nodes you can select actions like flash firmware or reset the node.
Can you confirm me that is it ok ?
Best regards.

fsaintma, 2018-09-28:
Ok, we don't have this behaviour with Chrome/Linux. 
Can you send us your chrome version ?
Should you test on another browser like Firefox ?
Can you see your "Chrome DevTools" and verify that is it ok, you don't have any errors ?
Thanks in advance.

npcode15, 2018-09-28:
Yes, I already did, please look at the comment above. It worked on Firefox.

Chrome version - Version 69.0.3497.100 (Official Build) (64-bit)
DevTools show everything as normal. DevTool console screenshot:
[image]

bformet, 2018-09-28:
I cannot reproduce this behavior with the exact same browser on windows 10.
Buttons should appear as soon as the experiment owner is the same as currently logged in user.
Can you give steps to reproduce?

npcode15, 2018-09-28:
Actually, there are no steps. I just log in and create an experiment. That's all.
I also tried in the Incognito mode but same result.

iotlovers, 2019-06-02:
i have the same problem
i tried linux (chrome and firefix) and win 10 (cherome and win 10)
what should i do ?
i checked by diffrent systems
[image]

vandaele, 2019-06-06:
That bug appeared when user log in with email address.
We deployed a bug fix yesterday.
You should log in / log out, and be able to find the actions buttons.

closed 2019-06-06 by vandaele