iot-lab/iot-lab #245: Multihop communication

teaserrahul, 2018-10-10:
I want to perform communication between two arbitrary nodes (either on wsn430 or M3). Basically, I want to implement a cluster-based aggregation, where some nodes will send their sensed data (temperature, light, etc.) to their corresponding cluster-head. CH should receive the data and display on the SSH frontend.

Please provide me with the source code regarding that.
Please ask if I am not clear.

rienafairefr, 2018-10-10:
Hi @teaserrahul,
You'd have to code your application in an operating system that can be deployed on IoT-LAB nodes, you can see in https://www.iot-lab.info/tutorials/ for some examples with RIOT and Contiki.
You'd have to choose/code your clustering approach, how the nodes communicate to elect a cluster head, how the nodes relay messages intended for the sink, stuff like that. Some things already exist to bootstrap, you could look into RPL, the organisation protocol developed for the [TRAXENS](https://hal.inria.fr/hal-01057738/file/ADHOCNETS_Presentation.pdf) solution, and a probable multiple others. The subject is really broad and an active area of research. You need to do a literature review of what you want and what has been already studied.

We provide infrastructure for you to do research on Wireless Sensor Network, we can't really provide you any source code apart from the short tutorials I linked above.

Best,
Matthieu Berthomé
FIT IoT-LAB team

teaserrahul, 2018-10-10:
Thanks @rienafairefr for the reply.
Please help me in getting output through monitoring profile. I deployed a simple code on wsn430 nodes and updated a monitoring profile as well. Now, how could I get the results regarding energy consumption and all that stuff?

rienafairefr, 2018-10-10:
WSN430 nodes were getting obsolete and the monitoring results were not good, so monitoring for WSN430 have been disabled. You should use M3 nodes, look at the tutorial [here](https://www.iot-lab.info/tutorials/monitoring-consumption-m3/)

teaserrahul, 2018-10-11:
[image]
 Please Help..!!!
I am getting this error while compiling the project in RIOT.

aabadie, 2018-10-11:
Hi @teaserrahul,

Can you provide your arm-none-eabi-gcc version ?
```
arm-none-eabi-gcc --version
```
If you are using the default 4.9.3, that might be the problem. See this [wiki page](https://github.com/iot-lab/iot-lab/wiki/Versions-of-the-GNU-toolchain-for-ARM) to use a newer (and supported by RIOT) one.

teaserrahul, 2018-10-11:
@aabadie Thanks a lot..!!!! It worked on changing the version. Thanks once again.
Please let me know, how can I add monitoring profile to the m3 nodes of my experiment through SSH frontend. I need to observe the energy consumption of m3 nodes.

teaserrahul, 2018-10-11:
@aabadie Thanks for your reply. 
The link that you suggested, it does not mention that how to add the created profile on the nodes. I have created a monitoring profile successfully but I am facing trouble in adding this profile to all the nodes in the experiment.

Since an option of adding monitoring profile occurs when adding a new experiment on the web portal. But if monitoring profile was not added at that time, then how we can add that after launching the experiment?

aabadie, 2018-10-11:
> But if monitoring profile was not added at that time, then how we can add that after launching the experiment?

You can't set monitoring to a running experiment. It must be defined when configuring the experiment.

teaserrahul, 2018-10-11:
@aabadie @rienafairefr Thank you for your generous replies..! Please resolve one more issue.

How to close the connection after accessing a node using "nc" command? Since serial_aggregator does not work for that node which was accessed through "nc" command. "serial_aggregator" shows connection close/refused error for that particular node.

What is the correct way to deal with this situation?

aabadie, 2018-10-11:
> What is the correct way to deal with this situation?

If you mean closing the nc client, then just ctrl+c is the way to go.

teaserrahul, 2018-10-11:
[image]
I am getting this error while trying to plot the graph of energy consumption by using the command: 
plot_oml_consum -p -i ~/.iot-lab/<experiment id>/consumption/m3-<id>.oml

teaserrahul, 2018-10-12:
> > What is the correct way to deal with this situation?
> 
> If you mean closing the nc client, then just ctrl+c is the way to go.

I am using ctrl+c to exit a node (which was accessed using "nc" command), but still, serial_agregator shows "connection closed" for those accessed nodes. Moreover, I am not able to access the same node again through "nc" command.

aabadie, 2018-10-12:
how do you launch nc ? If you add `&`, then the initial nc is still running.
If the SSH connection is aborted, your running process are stopped, so there shouldn't be any dangling connection in that case unless you start them in a screen.

closed 2019-06-17 by schrein