iot-lab/iot-lab #246: Node Deployment Failure

npcode15, 2018-10-10:
Hello, can someone please explain, why this happened.

**Site: Grenoble**
[image]

I tried restarting, submitting a fresh experiment. I tried deploying with firmware and without firmware, but, nothing seems to be working. Thanks.

hieu1204, 2018-10-11:
The same happened to me

rienafairefr, 2018-10-11:
I found the experiment in error, apparently our site-manager has an error creating the "sniffer" directory in your .senslab experiment directory. We're looking into it

```
2018-10-10 19:35:29,774 (ExperimentDao.java:285) Create /senslab/users/npandey/.senslab/137086//sniffer directory : mkdirs not allowed
2018-10-10 19:35:29,775 (ExperimentsResource.java:145) Start experiment 137086 results : 0 = [] 1 = [m3-108.grenoble.iot-lab.info, m3-107.grenoble.iot-lab.info, m3-110.grenoble.iot-lab.info, m3-109.grenoble.iot-lab.info, m3-106.grenoble.iot-lab.info, m3-105.grenoble.iot-lab.info]
```

npcode15, 2018-10-11:
@rienafairefr Thank you. The nodes deployed successfully.

closed 2018-10-11 by npcode15