iot-lab/iot-lab #249: Connection Closed

npcode15, 2018-10-18:
Hello,

I am getting the following error even though the experiment is still running.
![image](https://user-images.githubusercontent.com/25517038/47166004-da21c480-d2c0-11e8-94d8-
4b1922356cb5.png)

This has happened to me in the past as well

**Current Experiment:** 138368

Thank you

aabadie, 2018-10-18:
Instead of pasting images in issues, use Markdown code blocks syntax: surround the issue with triple back quotes. Same for commands.
For us, it makes life easier because we can copy-paste the snippets to reproduce your issues. You've already got a similar request in one of your RIOT issues. Please try to follow this recommendation.

In this issue, the image link is broken by the way.

npcode15, 2018-10-18:
@aabadie isn't this an IoTLAB issue?

I am sure I do not have any issue related to this in RIOT issues.

Yes, link was broken but I fixed the link already. 

'''
1539876881.700748;m3-99;Connection closed
1539876881.701361;m3-99;Connection closed
1539876881.701711;m3-99;Connection closed
1539876881.701994;m3-104;Connection closed
1539876881.702281;m3-104;Connection closed
1539876881.702533;m3-104;Connection closed
1539876881.702795;m3-100;Connection closed
1539876881.703073;m3-100;Connection closed
1539876881.703316;m3-100;Connection closed
1539876881.703581;m3-101;(<class 'socket.error'>, error(111, 'Connection refused'), <traceback object at 0x7f768e9c06c8>)
1539876881.703846;m3-102;(<class 'socket.error'>, error(111, 'Connection refused'), <traceback object at 0x7f768e9c0710>)
1539876881.704104;m3-103;(<class 'socket.error'>, error(111, 'Connection refused'), <traceback object at 0x7f768e9c06c8>)
1539876881.704406;m3-103;Connection closed
1539876881.704688;m3-103;Connection closed
1539876881.704931;m3-103;Connection closed
1539876881.705186;m3-101;Connection closed
1539876881.705514;m3-101;Connection closed
1539876881.705780;m3-101;Connection closed
1539876881.706040;m3-102;Connection closed
1539876881.706311;m3-102;Connection closed
1539876881.706548;m3-102;Connection closed
1539876881.706775;Loop finished, all connection closed
'''

aabadie, 2018-10-18:
The syntax is triple **back** quotes not regular single quotes.

npcode15, 2018-10-18:
I am sorry. I do use them for codes.

aabadie, 2018-10-18:
> I do use them for codes.

It is the same syntax. For code, you can specify the language Python, C, etc to use a nice syntactic color.

npcode15, 2018-10-18:
Sure. I will keep that in mind. Thanks.

And regarding this, restarting the experiment is the only solution?

aabadie, 2018-10-18:
> restarting the experiment is the only solution?

Normally not. Does it work if you stop the serial_aggregator and connect to each node separately using nc ?

npcode15, 2018-10-18:
Actually, Serial Aggregator went back to normal. I did not change anything.

npcode15, 2018-10-22:
@aabadie 

Please help me resolve this issue. I keep facing this issue intermittently, initially, this was occurring at the Grenoble site only but I started facing this issue at the Paris site as well. The only option is to restart the experiment. But, I have to wait if someone else requests the resources before me. 

Thank you.

aabadie, 2019-06-17:
I think the connection issues has been fixed recently. I'll close this, but feel free to reopen if you still have issues.

closed 2019-06-17 by aabadie

robertobruzzese, 2021-07-26:
The problem is now again in grenoble