iot-lab/iot-lab #251: Makefile / compile errors with gnrc

rboyer-at-brksd-com, 2018-10-27:
I'm sorry for such a basic question, but I cannot get the gnrc_networking examples to compile on either the Saclay or Grenoble sites.  I'm following the tutorial instructions carefully, and have been able to compile RIOT-OS on my personal Ubuntu server and play around with Native.  I'm trying to work with real nodes now.  All help appreciated!  (I'm a fairly basic Linux user, but a careful follower of directions.)

- Robert BOYER

"make: Entering directory '/senslab/users/boyer/riot2017.07/RIOT/examples/gnrc_networking'
Building application "gnrc_networking" for "iotlab-m3" with MCU "stm32f1".

cc1: error: -I/senslab/users/boyer/riot2017.07/RIOT/core/include: No such file or directory [-Werror]
In file included from /senslab/users/boyer/riot2017.07/RIOT/examples/gnrc_networking/main.c:23:0:
/senslab/users/boyer/riot2017.07/RIOT/sys/include/shell.h:25:28: fatal error: kernel_defines.h: No such file or directory
 #include "kernel_defines.h"
                            ^
cc1: all warnings being treated as errors
compilation terminated.
/senslab/users/boyer/riot2017.07/RIOT/Makefile.base:83: recipe for target '/senslab/users/boyer/riot2017.07/RIOT/examples/gnrc_networking/bin/iotlab-m3/application_gnrc_networking/main.o' failed"

aabadie, 2018-10-28:
It seems like a compiler version issue. RIOT requires the latest release of arm-none-eabi-gcc, 7.3, and the default version on IoT-LAB is 4.9.

Can you try the following command:
```
$ source /opt/riot.source
```
and try to build again ?

Otherwise, there's the IoT-LAB contain [a wiki page](https://github.com/iot-lab/iot-lab/wiki/Versions-of-the-GNU-toolchain-for-ARM) detailing the available gcc toolchains.

rboyer-at-brksd-com, 2018-10-28:
Hi aabadie - that worked perfectly, thank you so much!  I was able to compile the example grnrc_networking application.  You have my gratitude!

closed 2018-10-28 by rboyer-at-brksd-com

rboyer-at-brksd-com, 2018-10-29:
Alexandre,

Yes, that fixed the problem, everything compiles fine now. Thank you very, very much!

Robert BOYER
President
Consulting on People   -   Products   -   Processes   -   Technology

See how we are different:        https://vimeo.com/185065043

Review our services:               http://www.siliconharbor.services

USA Eastern Time (UTC-5)
+1.843.972.9284<tel:+1.843.972.9284>  | rboyer@siliconharbor.services

Disclaimer: This e-mail message and attachments may contain confidential information intended only for named recipient(s). If you are not the intended recipient, you should not review, disseminate, distribute or copy this e-mail. Please notify the sender immediately and delete this e-mail.