iot-lab/iot-lab #252: Node Deployment Failure

npcode15, 2018-11-08:
Hello, 

I am having issues in deplying nodes to carry on experiment. Nodes fail to deploy successfully.

Experiments IDs: #141205 to #141209
Site: Grenoble

This started 2-3 days ago when nodes m3-11 and m3-12 failed to deploy (site: paris). 
Today nodes 98-99/101-102 failed to deploy (site: grenoble)

This has happened with me in the past as well (Issue #246).

Please have a look into this issue. Thank you.

rienafairefr, 2018-11-09:
Hello Navneet,
yes, unfortunately the deployment is not a 100% certain process, we recommend that your experiment  be able to be tolerant to some amount of failure among the nodes. Unless you absolutely need specific nodes, you should design your experiment so that it can be deployed to many different parts of the testbed. The nodes are identical and run exactly the same software. And considering the deployment in a grid for almost all the sites, the environment is not markedly different.
We could look into the`empty node list` if we knew the experiment id.
Best,
Matthieu

npcode15, 2018-11-09:
Hello Matthieu,

I am afraid I do not remember the '''empty node list''' experiment.

My experiment specifically use some IPs. I have a list of about 6-8 nodes which I usually select. In worst scenario, I deploy to a new node and get the IP.

This is the reason I haven't been able to use WSN430 nodes. I did open another issue previously which hasn't been answered yet. I would really appreciate it if you could answer that issue as well.

But, as of now I the nodes are working fine.

Thank you.

aabadie, 2019-06-17:
Looks like this is working now (according to the edit in the initial comment), let's close this.

closed 2019-06-17 by aabadie