iot-lab/iot-lab #254: CoAP and Border router issue with TSCH

dverasqrz, 2019-03-23:
Dear all,

I've been studying the IoT-Lab code to provide CoAP with RPL and TSCH in my home network.
The problem is I could not submit GET and POST methods to the CoAP server from the computer. Neither the CoAP server nor the border router receive ping from the computer.

I put both motes to communicate, but I have faced these issues that I couldn't solve yet.

When using the standard configuration of 05-rpl-tsch-border-router and 06-rpl-tsch-coap, I can receive just one ping reply from the border router to the computer, right after pushing the restart button of the border router, and then it loses the connection with the computer.
However, both nodes (CoAP server and Border router) seem to keep on communicating with each other, as described below:

user@contikiU8:~/contiki/examples/DIEGO-EXP/BR$ sudo ~/contiki-ng/tools/serial-io/tunslip6 -v2 -t tun0 -s /dev/ttyUSB1 fd00::1/64
********SLIP started on ``/dev/ttyUSB1''
opened tun device ``/dev/tun0''
ifconfig tun0 inet `hostname` mtu 1500 up
ifconfig tun0 add fd00::1/64
ifconfig tun0 add fe80::0:0:0:1/64
ifconfig tun0

tun0: flags=4305<UP,POINTOPOINT,RUNNING,NOARP,MULTICAST>  mtu 1500
        inet 127.0.1.1  netmask 255.255.255.255  destination 127.0.1.1
        inet6 fd00::1  prefixlen 64  scopeid 0x0<global>
        inet6 fe80::106a:6b52:2719:69c7  prefixlen 64  scopeid 0x20<link>
        inet6 fe80::1  prefixlen 64  scopeid 0x20<link>
        unspec 00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00  txqueuelen 500  (UNSPEC)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

*** Address:fd00::1 => fd00:0000:0000:0000
Got configuration message of type P
Setting prefix fd00::
TSCH-schedule: add_slotframe 0 3
TSCH-schedule: add_link 0 15 1 0 0 255
TSCH: starting as coordinator, PAN ID cdef, asn-0.0
TSCH: starting as coordinator
Server IPv6 addresses:
 fd00::212:4b00:18e0:ba05
 fe80::212:4b00:18e0:ba05
TSCH-queue: packet is added put_index=0, packet=#0x20000654
TSCH: enqueue EB packet 37 18
TSCH-queue: packet is removed, get_index=0
TSCH-queue: packet is added put_index=0, packet=#0x20000654
TSCH: send packet to 255 with seqno 0, queue 0 1, len 16 96
TSCH-queue: packet is removed, get_index=0
TSCH-queue: packet is added put_index=1, packet=#0x20000654
TSCH: enqueue EB packet 37 18
TSCH-queue: packet is removed, get_index=1
TSCH-queue: packet is added put_index=1, packet=#0x20000654
TSCH: send packet to 255 with seqno 0, queue 0 1, len 16 96
TSCH-queue: packet is removed, get_index=1
TSCH-queue: packet is added put_index=2, packet=#0x20000654
TSCH: enqueue EB packet 37 18
TSCH-queue: packet is removed, get_index=2
TSCH: received from 180 with seqno 65535
TSCH: received from 180 with seqno 1
TSCH: received from 180 with seqno 65535
TSCH: received from 180 with seqno 2
TSCH-queue: packet is added put_index=2, packet=#0x20000654
TSCH: send packet to 255 with seqno 0, queue 0 1, len 16 96
TSCH-queue: packet is removed, get_index=2

**However**, when I try to see the routing table of border router, nothing appears, i.e. the border router does not communicate with the computer:

user@contikiU8:~/firefox$ lynx -dump http://[fd00::212:4b00:18e0:ba05]
...

Reviewing the TSCH configuration code (tsch-project-conf), I found that when I comment the line:
//#define RPL_CALLBACK_NEW_DIO_INTERVAL tsch_rpl_callback_new_dio_interval
the border router can now reply the ping command, however, it cannot communicate with the CoAP server:

**Reply OK from the border router to the computer**:
user@contikiU8:~/firefox$ ping6 fd00::212:4b00:18e0:ba05
PING fd00::212:4b00:18e0:ba05(fd00::212:4b00:18e0:ba05) 56 data bytes
64 bytes from fd00::212:4b00:18e0:ba05: icmp_seq=1 ttl=64 time=31.3 ms
64 bytes from fd00::212:4b00:18e0:ba05: icmp_seq=2 ttl=64 time=22.0 ms

**Without communication between the motes**:
user@contikiU8:~/contiki/examples/DIEGO-EXP/BR$ sudo ~/contiki-ng/tools/serial-io/tunslip6 -v2 -t tun0 -s /dev/ttyUSB1 fd00::1/64
********SLIP started on ``/dev/ttyUSB1''
opened tun device ``/dev/tun0''
ifconfig tun0 inet `hostname` mtu 1500 up
ifconfig tun0 add fd00::1/64
ifconfig tun0 add fe80::0:0:0:1/64
ifconfig tun0

tun0: flags=4305<UP,POINTOPOINT,RUNNING,NOARP,MULTICAST>  mtu 1500
        inet 127.0.1.1  netmask 255.255.255.255  destination 127.0.1.1
        inet6 fd00::1  prefixlen 64  scopeid 0x0<global>
        inet6 fe80::3193:1dac:dd12:99d  prefixlen 64  scopeid 0x20<link>
        inet6 fe80::1  prefixlen 64  scopeid 0x20<link>
        unspec 00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00  txqueuelen 500  (UNSPEC)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

*** Address:fd00::1 => fd00:0000:0000:0000
Got configuration message of type P
Setting prefix fd00::
TSCH: scanning on channel 15
TSCH: starting as node
Server IPv6 addresses:
 fd00::212:4b00:18e0:ba05
 fe80::212:4b00:18e0:ba05
TSCH:! not associated, drop outgoing packet
TSCH: scanning on channel 26
TSCH: scanning on channel 15
TSCH: scanning on channel 25
TSCH: scanning on channel 20
TSCH: scanning on channel 15
TSCH: scanning on channel 25
TSCH: scanning on channel 15
TSCH:! not associated, drop outgoing packet
TSCH: scanning on channel 25
TSCH: scanning on channel 15
TSCH: scanning on channel 25
TSCH: scanning on channel 26
...
...

Could anyone please help me with this issue?

Many thanks

dverasqrz, 2019-03-24:
I have an update...

Sometimes, the message is as follows:

TSCH: scanning on channel 25
TSCH: scanning on channel 15
TSCH: scanning on channel 25
TSCH: scanning on channel 26
**TSCH: association: received packet (0 bytes) on channel 26
TSCH:! parse_eb: failed to parse frame
TSCH:! failed to parse EB (len 0)**
TSCH: scanning on channel 15
TSCH: scanning on channel 25
TSCH: scanning on channel 20

schrein, 2019-06-17:
Hello @dverasqrz, thank you for trying FIT IoT-LAB tutorials. In order to debug your own setup with your own hardware, we advise you to directly ask your questions to Contiki developers.

closed 2019-06-17 by schrein