iot-lab/iot-lab #268: common.py:147: ConversionWarning

mingridefeng, 2019-06-04:
hi, 
I am doing the "Monitor the consumption of M3 node during an experiment" experiment, when I tried to use this command line, <login>@grenoble:~$ plot_oml_consum -p -i ~/.iot-lab/<experiment id>/consumption/m3-<id>.oml
I met some issues as below: 
<login>@grenoble:~$ plot_oml_consum -p -i ~/.iot-lab/171085/consumption/m3_100.oml
/usr/local/lib/python2.7/dist-packages/oml_plot_tools/common.py:147: ConversionWarning: Some errors were detected !
    Line #2128 (got 7 columns instead of 8)
  invalid_raise=False)

thanks for the help.

rienafairefr, 2019-06-17:
Hi,
This is a normal and likely warning, because the OML files are written in chunks, so usually the last line is missing some columns. This is just a warning and should not abort anything. 
I ran the plot_oml_consum command you wrote with the specific file you mentioned, and it worked fine (graph below)
[image]

closed 2019-06-17 by rienafairefr

mingridefeng, 2019-06-20:
> Hi,
> This is a normal and likely warning, because the OML files are written in chunks, so usually the last line is missing some columns. This is just a warning and should not abort anything.
> I ran [...]

Hi, 
Thank you so much for the replying, probably the main issue was I use MacOS to connect with testbed, this experiment suggested us using X11Forwarding which means "ssh -X", there are some specific setting for X11Forwarding on MacOS.