iot-lab/iot-lab #272: question about RPL / latency

yamini4real, 2019-06-24:
i want to calculate the latency bitween source and destination node.
i did some tutorials but i didnt see anything about latency and i couldnt measure latency and there were no option to o create a profile for monitoring consumption for latency can antone help me ?

yamini4real, 2019-08-01:
why no one answears?

rienafairefr, 2019-08-01:
Hi @yamini4real

Sorry about delayed answers.

There is no built-in monitoring of network latency. In IoT-LAB we strive to give you a lot of freedom to run whatever you want, and we won't constrain people to use a certain routing/OS/application, so it makes little sense to build something to measure latency between source and destination node, that will only be useful for a certain type of applications/protocol/certain users.

Using the tools we provide, you should be able to build something to measure the latency yourself.
- when a packet is sent by the source, get the time. You need to either you look at the time the packet was output on the serial console, read by a serial_aggregator -simplest but less accurate-, or you query the control node for its time using I2C -more complicated but more accurate
- when a packet is received at the destination, get the time (same as the source)
you get the latency by making a diff between the two.
Usually, people run a serial_aggregator on their experiment, output the packets on the serial link of each node, then when the experiment is finished, process the log output using their language of choice.

Make note: Because we are synchronising the times between the different nodes through the network, and through USB-serial links, don't expect a µs-level accuracy. More like a few ms

Matthieu