iot-lab/iot-lab #273: Issue with Border Router and Tunslip

Rjlmota, 2019-07-29:
I had a few scripts that ran all my experiments automatically. But, they suddenly stopped working. I tried to run an experiment manually and discovered that that the tunslip was giving an error when the border router firmware was updated.
error:

tunslip6: serial_to_tu: write: Invalid argument
ifconfig tun0 down
netstat -nr | awk '{ if ($2 == "tuns0") print "route delete -net "$1; }' | sh

I asked my friend to create a new account, so  I could follow this tutorial:
https://www.iot-lab.info/tutorials/contiki-private-ipv6-m3/

But I got the same error, even though I was in a new environment.

I honestly don't know if I'm missing something obvious or what. Can someone help me?

fsaintma, 2019-07-31:
Hi,

You are right I can reproduce the error. Meanwhile we find a workaround you should switch the steps and it will work perfectly:

1°) flash the border router firmware
2°) launch the tunslip command after

Sorry for the inconvenience.

Best regards.

Rjlmota, 2019-08-05:
It does continue running when I do this. But, the routes are not created properly, I believe. I'm using 20 nodes and the border router creates only 8 of those most of the time. I don't know if the problem is due to the order of the commands, though.

fsaintma, 2019-08-05:
Hi,
We have found the reason of the bug (debug print log attempt to write on the TUN interface)  and we will find a workaround. For your BR and routes creation it's not correlated. Maybe a problem of radio neighborhood. Are you sure that all your experiment nodes should communicate together. Maybe you should also do some tuning Contiki configuration.
Hope that help you.