iot-lab/iot-lab #276: Can't create a new user?

mmaxus35, 2019-08-22:
I Can't sing up your testbed. It gives this error.  email is my academic email.

I need to use a node to test my Riot OS program.

Thanks.

[image]

vandaele, 2019-08-23:
You already have an account active since October 2018.

The "Forget your password?" link will bring you to [this page](https://www.iot-lab.info/testbed/reset), where you could reset your password by giving your email address.

mmaxus35, 2019-08-23:
Dear @vandaele , My e-mail address is closed permanently because of my research project is ended with the university, so we are in new research based on RIOT and new e-mail address is assigned. I tried to different e-mails just to be sure that there are any problems with e-mail format. Moreover, for the project, my friend requires an account too. 

As a result, 
i) I can't reset my password because name.surname@yasar.edu.tr is not accessible by me anymore.
ii) I tried to get a new account on my other academic e-mail 91170000255@ogrenci.ege.edu.tr.

Is there any chance you can edit my e-mail address burak.karaduman@yasar.edu.tr to 91170000255@ogrenci.ege.edu.tr , or can I get a new account using my accessible e-mail 91170000255@ogrenci.ege.edu.tr ?

Thanks.

vandaele, 2019-08-23:
I did change your email address in your existing account. You should be able to reset your password now.

mmaxus35, 2019-08-23:
Thank you so much.

şunu yazdı:

closed 2019-08-23 by vandaele