iot-lab/iot-lab #277: TSCH association is taking long time in FIT IoT lab

Alakesh1025, 2019-11-15:
I am trying to simulate Contiki NG based TSCH on M3 node in FIT IoT lab but node are not joining the network though I have placed several coordinator node in between the nodes. What to do?

rienafairefr, 2019-11-18:
Hi, 
TSCH in contiki-ng support on the IoT-LAB M3 nodes is still broken unfortunately, and we're not sure where the problem lies. We preferred to ship a partial Contiki-NG support, instead of not shipping any support at all.
If you manage to pinpoint where the problem lies, do share it so that we can send it upstream.

FIT IoT-LAB team

fsaintma, 2019-11-18:
Hi @Alakesh1025 and @rienafairefr  
The IoT-LAB contiki-ng TSCH support is now operational with the latest release 4.4. If you use the 6tisch simple node example don't forget to use on the coordinator shell the following command:  rpl-set-root 1. It will also automatically activate 6tisch coordinator (tsch-set-coordinator 1).
Best regards.

Alakesh1025, 2019-11-18:
The problem is with node association. The PAN coordinator keeps on sending its control packets (EB, DIO) and the new nodes keep on scanning for control packet.

@fsaintma  Sir, I used the latest contiki 4.4 and also used rpl-set-root 1 and tsch-set-coordinator 1 commands but still node are not joining the network.

fsaintma, 2019-11-18:
@Alakesh1025
 
I have just tested a 6tisch network (examples/6tisch/simple-node) on IoT-LAB Grenoble site with 10 M3 nodes and one coordinator. I can see the RPL neighbors and ping them. So what is your problem of node association ?

```
tsch-status
TSCH status:
-- Is coordinator: 1
-- Is associated: 1
-- PAN ID: 0x81a5
-- Is PAN secured: 0
-- Join priority: 0
-- Time source: none
-- Last synchronized: 0 seconds ago
-- Drift w.r.t. coordinator: 0 ppm
-- Network uptime: 6966 seconds

rpl-nbr
RPL neighbors:
fe80::9181  256,   128 =>   384 --  1   a   (last tx 0 min ago)
fe80::a072  256,   128 =>   384 --  1   a   (last tx 0 min ago)
fe80::a071  256,   140 =>   396 --  4   af  (last tx 0 min ago)
fe80::9382  256,   128 =>   384 --  1   a   (last tx 0 min ago)
fe80::b080  256,   128 =>   384 --  1  ba   (last tx 0 min ago)
fe80::8477  396,   256 =>   652 --  0   a   (no tx)
fe80::9881  256,   224 =>   480 --  1   a   (last tx 0 min ago)
fe80::a881  256,   224 =>   480 --  1   a   (last tx 0 min ago)
fe80::1062  422,   128 =>   550 --  0   a   (no tx)

ping fe80::9181
Pinging fe80::9181
Received ping reply from fe80::9181, len 4, ttl 64, delay 50 ms

```

Alakesh1025, 2019-11-18:
@fsaintma  Sir, I have used 16 channels and also ran the for only 1hr. Let me check with 4 channel and less time

Alakesh1025, 2019-11-18:
Sir, it is working fine with 16 channels. Sir I have last few questions
1. How to get the association time( the time when a node receives its first EB)? Should I use print command? I have turn off the log file because it was interrupting to send command. 
2. Also I have to count the number of EB sent and receives. Should I use a counter variable and print its value?

@fsaintma @rienafairefr  Thank you Sir

Alakesh1025, 2019-11-23:
@fsaintma  Could you please tell me how to save the log files? I am not able to log files though output is showing in web console.

fsaintma, 2019-11-26:
Hi @Alakesh1025 
Sorry for my late answer. I suppose that you use the IoT-LAB webportal to view the logs ?
During an experiment the M3 node serial port (what you see in the web console) is forwarded on the IoT-LAB frontend SSH (port 20000)
So if you want to store some logs print on the serial port you must connect to the IoT-LAB frontend SSH site and read the serial port (ex: nc m3-<id> 20000) and save it in a file in your home.
You can also view this tutorial which explain you how to read/write the experiment node's serial port
in the same time: https://www.iot-lab.info/tutorials/serial-aggregator/
Best regards.