iot-lab/iot-lab #278: [Rest API] Grenoble : "Script execution not supported on this site"

jeremydub, 2020-02-03:
Hi,
I am not able anymore to launch frontend shell scripts in Grenoble using the API. The debug log "run_script.log" only says "Script execution not supported on this site".

It was working fine previous weeks/months, until now. I tried by setting the script during the experiment submission (POST /experiments) and also by setting the script via POST /experiments/ID/scripts/run.

Maybe it is a coincidence but it also seems like the backend changed recently since language has been set to English (instead of French) and some commands I was using are not present anymore (eg. 'bc' command)

fsaintma, 2020-02-04:
Hi @jeremydub. Thanks for your bug reporting. You are right the run script manager didn't work following the server maintenance. We have migrated the Debian distribution (Jessie -> Stretch) on the IoT-LAB Grenoble frontend server. In the first place we have added bc package and don't hesitate to communicate us if others are missing. For the run script manager (bug regression due to a new gestion of tmpfs filesystem storage and noexec option) we  have done a patch but we must  wait that no experiments is running on the Grenoble site before deploying it. It seems that currently you are running some experiments :)
We keep you informed when it's done.
Best regards.

jeremydub, 2020-02-04:
Great ! Happy to hear it was a scheduled maintenance :) Thank you for maintaining the system !
Jeremy