iot-lab/iot-lab #281: npm installation fails

archanabhat1990, 2020-02-18:
I am working with CoAP experiment where I need to install npm.  when I install npm and enter the password it fails. Please help me with the right password for the installation.

vandaele, 2020-02-18:
why do you need to install npm to work with CoAP ?

archanabhat1990, 2020-02-18:
Hi
Thanks for the response.
https://www.iot-lab.info/tutorials/contiki-coap-m3/   . Running this
example experiment. When I execute 16th step, CoAP not found. So when I saw
about CoAP-CLI those are the request to execute GET, PUT POST and DELETE.
So how to install it ?

Regards,
Archana Bhat

vandaele, 2020-02-21:
We had an issue with the coap-cli nodejs package installed on our new Debian Stretch servers.
We removed our dependency to nodejs and installed another CoAP client on frontend servers, e.g. aiocoap-client.
I still have to modify the tutorials.

closed 2020-02-21 by vandaele

archanabhat1990, 2020-02-22:
Thank you for the reply. When can this issue be solved completely ?

Regards,
Archana Bhat

archanabhat1990, 2020-02-24:
Any help to modify/work with respect to CoAP, feel free to ask me.

Regards,
Archana Bhat

vandaele, 2020-02-25:
All the tutorials of our website, specially instructions using CoAP client, have been changed on the IoT-LAB website.