iot-lab/iot-lab #283: Version of Pycom Fipy @ Saclay

EGo41000, 2020-04-05:
I am trying to test Lora Mesh on Pycom Fipy at Saclay.
I follow https://docs.pycom.io/firmwareapi/pycom/network/lora/
When trying 
`pymesh = lora.Mesh()`
But get 
`AttributeError: 'LoRa' object has no attribute 'Mesh'`
According to https://github.com/pycom/pycom-libraries/issues/66 device should be upgraded to 1.19.0.b4 (or latest)
Saclay version are 
`Pycom MicroPython **1.18.2.r2** [v1.8.6-849-5cf02dc] on 2019-03-07; FiPy with ESP32`
Could Fipy devices be upgraded ?
Thanks,
Emmanuel.

aabadie, 2020-04-06:
Thanks for reporting this problem. To upgrade the firmware that is running on the Pycom boards, we need a physical access, since this cannot be done remotely.
We'll do that when the lockdown is over.

I'm surprised that you were able to use them anyway since we never provided any documentation about how to use them on IoT-LAB :)

EGo41000, 2020-04-06:
Hello Alexandre,
Thanks for your support !
I already have some Pycom products (Lopy in particular). I was wondering if I could prog them (in C), but in fact they are only running micropython, which is 'simple' to use.
Are you planning possibility to flash them (from C) like other platforms ? 
BR,
Let me know when you succeed to upgrade.
Emmanuel.

HADJMOHAND, 2024-11-15:
Hello Emmanuel,
I saw your message regarding testing LoRa Mesh on Pycom FiPy at Saclay and the firmware update issue. I hope you find a solution to upgrade the devices.

I am also working with LoRa on IoT-LAB, but I am not sure where to program the node to make it work properly. Could you please guide me on where and how to program a node that uses LoRa in IoT-LAB? Any advice or help would be greatly appreciated.

Thank you in advance for your assistance!

Best regards,
Amel