iot-lab/iot-lab #284: Account Registration

herringm811, 2020-04-07:
Having an issue trying to register for an account. I get an error when submitting. All it says is user account error.

Sdeshmukh2020, 2020-04-20:
Having an issue trying to register for an account. I get an error when submitting. All it says is user account error.

priyeashbala, 2020-04-21:
Having an issue trying to register for an account. I am not getting the activation e-mail to my university e-mail inbox. It says that I should activate my account but I did not get any e-mail.

ddujovne, 2020-05-28:
I have tried to reset my password but the button does not work (I know my account is active, I cannot register it again). I do not receive any password reset mail at all.

rilanani, 2020-11-05:
Did anyone resolve this issue?

rilanani, 2020-11-05:
Anyone was able to solve this?? Mine shows error during signup create user error.