iot-lab/iot-lab #286: Plotting power consumption diagram issue

sabriCHALLOUF, 2020-11-01:
Hi,
I detected an issue when plotting the power consumption graph using plot_oml_consum.

Test Case : 
     + BOARD : M3
     + Binary : binary of the example timer_periodic_wake_up (from the RIOT OS sources)
     + Number of nodes : 4
     + IoTLab Site : Grenoble
     + Used nodes : 95, 96, 101 and 102
     + Experiments numbers : 234396, 234385 and 234384

Expected Results : 
     + OML files contains power consumption values 👍 
     + OML plotter should use the OML file values to plot power consumption graph.

Real Results : 
     + OML plotter ( plot_oml_consum -p -i ~/.iot-lab/last/consumption/m3_95.oml) detect some errors when lunched 👎 

Here is the errors message : 
===================================================================================
challouf@grenoble:~$ plot_oml_consum -p -i ~/.iot-lab/last/consumption/m3_95.oml
/usr/local/lib/python2.7/dist-packages/oml_plot_tools/common.py:147: ConversionWarning: Some errors were detected !
    Line #2106 (got 7 columns instead of 8)
  invalid_raise=False)
Traceback (most recent call last):
  File "/usr/local/bin/plot_oml_consum", line 24, in <module>
    oml_plot_tools.consum.main()
  File "/usr/local/lib/python2.7/dist-packages/oml_plot_tools/consum.py", line 153, in main
    consumption_plot(data, opts.title, selection)
  File "/usr/local/lib/python2.7/dist-packages/oml_plot_tools/consum.py", line 115, in consumption_plot
    oml_plot(data, title, [MEASURES_D[value]])
  File "/usr/local/lib/python2.7/dist-packages/oml_plot_tools/consum.py", line 137, in oml_plot
    plt.figure()
  File "/usr/local/lib/python2.7/dist-packages/matplotlib/pyplot.py", line 533, in figure
    **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/matplotlib/backend_bases.py", line 161, in new_figure_manager
    return cls.new_figure_manager_given_figure(num, fig)
  File "/usr/local/lib/python2.7/dist-packages/matplotlib/backends/_backend_tk.py", line 1046, in new_figure_manager_given_figure
    window = Tk.Tk(className="matplotlib")
  File "/usr/lib/python2.7/lib-tk/Tkinter.py", line 1823, in __init__
    self.tk = _tkinter.create(screenName, baseName, className, interactive, wantobjects, useTk, sync, use)
_tkinter.TclError: no display name and no $DISPLAY environment variable
==============================================================================

I like to add that I used also the -X option when logging with SSH but still the same problem.
Thank you in advance

fsaintma, 2020-11-03:
Hi, I have just tested with your homedir.  I put my SSH public key on your account profile and test to plot the file with success. I save the plot in your homedir with the name `Figure_1.png`

```
fsaintma@orval:~$ ssh -X challouf@grenoble.iot-lab.info
challouf@grenoble:~$ plot_oml_consum -p -i ~/.iot-lab/last/consumption/m3_95.oml
^Z
[1]+  Stopped                 plot_oml_consum -p -i ~/.iot-lab/last/consumption/m3_95.oml
challouf@grenoble:~$ ls
...  Figure_1.png  ...
```
So I suppose that you have a problem with your SSH client configuration and export display.

Best regards.

sabriCHALLOUF, 2020-11-07:
Hi,
Thank you for the response.
I updated my ssh keys and still the same problem.
I don't know what to do to resolve the problem.

Best Regards

sabriCHALLOUF, 2021-04-02:
I Like to add that the problem appears in the Windows environment but not in Linux one. I tried with Ubuntu 16 and 20 and it works well.

closed 2021-04-02 by sabriCHALLOUF