iot-lab/iot-lab #290: submit expiriment problem

souhilahana, 2021-01-24:
Hi iot-lab users,

i have a script client.py that i worked with it before one week without a problems, but when the iot-lab administrators upgraded python version, i got a lot of errors that i fixed it but, the script still blocked at this line run_command ("iotlab-experiment stop", True, True). i m not familiarized with python, any one help me to fix this problem?
thank you 

# Running the sh script
try:
    user, err = subprocess.Popen(shlex.split("whoami"), stdout=subprocess.PIPE, stderr=subprocess.PIPE).communicate()
    if err:
        raise Exception("Failed to query username:\n" + err + clr.ENDC)
    user = user[:-1]
    site, err = subprocess.Popen(shlex.split("uname -n"), stdout=subprocess.PIPE, stderr=subprocess.PIPE).communicate()
    if err:
        raise Exception("Failed to query IoT-Lab site:\n" + err + clr.ENDC)
    site = site[:-1]
    if not os.path.exists(".iotlabrc"):
        print clr.OKGREEN + clr.BOLD + "Logging in as {}@{}, please authentificate (note that password isn't displayed)".format(user, site) + clr.ENDC
        if run_command("iotlab-auth -u {}".format(user)):
            raise Exception("Failed to authentificate" + clr.ENDC)
    else:
        print clr.OKGREEN + clr.BOLD + "Already logged in as {}@{}".format(user, site) + clr.ENDC
    print clr.OKGREEN + clr.BOLD + "Stopping all experiments" + clr.ENDC
    run_command("iotlab-experiment stop", True, True)
    run_command("iotlab-experiment wait --state Finished,Error", True, True)
    if not args.cleanup:
        time.sleep(2)
        print clr.OKGREEN + clr.BOLD + "Retrieving IoT-Lab testbed status" + clr.ENDC
        json_out, err = subprocess.Popen(shlex.split("iotlab-experiment info --site {} -l".format(site)), stdout=subprocess.PIPE, stderr=subprocess.PIPE).communicate()
        if err:
            raise Exception("Error while retrieving IoT-Lab testbed status:\n" + err + clr.ENDC)
        items = json.loads(json_out)['items']
        items = [item for item in items if item['site'] == site]
        pNodes = [item for item in items if item['network_address'].startswith('{}-{}.'.format(args.architecture, args.source)) or item['network_address'].startswith('{}-{}.'.format(args.architecture, args.sink))]
        if len(pNodes) < 2:
            raise Exception("At least one node you selected is invalid" + clr.ENDC)
        pos = {}
        uid = {}
        for item in pNodes:
            if item['network_address'].startswith('{}-{}.'.format(args.architecture, args.source)):
                if item['state'] != 'Alive':
                    raise Exception("The source node ({}-{}) is {}".format(args.architecture, args.source, item['state'].lower()) + clr.ENDC)
                pos[str(args.source)] = (item['x'], item['y'])
                uid[item['uid']] = str(args.source)
            elif item['network_address'].startswith('{}-{}.'.format(args.architecture, args.sink)):
                if item['state'] != 'Alive':
                    raise Exception("The sink node ({}-{}) is {}".format(args.architecture, args.sink, item['state'].lower()) + clr.ENDC)
                pos[str(args.sink)] = (item['x'], item['y'])
                uid[item['uid']] = str(args.sink)
        # Query intermediate nodes
        if args.intermediate is not None:
            if not isinstance(args.intermediate, list):
                # By quantity
                count = int(args.intermediate[1:])
                args.intermediate = []
                for item in items:
                    if count <= 0:
                        break
                    if item['state'] == 'Alive' and not (item['network_address'].startswith('{}-{}.'.format(args.architecture, args.source)) or item['network_address'].startswith('{}-{}.'.format(args.architecture, args.sink))):
                        src = re.search('{}-(\d+).'.format(args.architecture), item['network_address'])
                        if src:
                            args.intermediate.append(src.group(1))
                            pos[str(src.group(1))] = (item['x'], item['y'])
                            uid[item['uid']] = str(src.group(1))
                            count -= 1
                if count > 0:
                    print clr.WARNING + clr.BOLD + "Warning: missing {} intermediate nodes".format(count) + clr.ENDC
                print clr.OKGREEN + clr.BOLD + "Found {} available nodes:".format(len(args.intermediate)) + clr.ENDC
                print clr.OKGREEN + ', '.join(args.intermediate) + clr.ENDC
            else:
                # By nodes list
[... 53 lines cut ...]
    print clr.FAIL + clr.BOLD
    traceback.print_exc()
    os._exit(1)
try:
    if args.intermediate is not None:
        json_out, err = subprocess.Popen(shlex.split("iotlab-experiment submit -n sensevidRPL -d {} -l {},{},{},{}{} -l {},{},{},{}{} -l {},{},{},{}{}".format(args.duration, site, args.architecture, args.source, "/senslab/users/{}/{}".format(user, src_file), ",{}".format(args.profile) if args.profile else "", site, args.architecture, args.sink, "/senslab/users/{}/{}".format(user, rcv_file), ",{}".format(args.profile) if args.profile else "", site, args.architecture, '+'.join(args.intermediate), "/senslab/users/{}/{}".format(user, itm_file), ",{}".format(args.profile) if args.profile else "")), stdout=subprocess.PIPE, stderr=subprocess.PIPE).communicate()
    else:
        json_out, err = subprocess.Popen(shlex.split("iotlab-experiment submit -n sensevidRPL -d {} -l {},{},{},{}{} -l {},{},{},{}{}".format(args.duration, site, args.architecture, args.source, "/senslab/users/{}/{}".format(user, src_file), ",{}".format(args.profile) if args.profile else "", site, args.architecture, args.sink, "/senslab/users/{}/{}".format(user, rcv_file), ",{}".format(args.profile) if args.profile else "")), stdout=subprocess.PIPE, stderr=subprocess.PIPE).communicate()
    if err:
        raise Exception("Error while submiting the experiment:\n" + err + clr.ENDC)
    id = json.loads(json_out)['id']
    if run_command("iotlab-experiment wait -i {}".format(id)):
        raise Exception("Failed, either it has timed out or the experiment creation failed" + clr.ENDC)
    print clr.OKGREEN + clr.BOLD + "Experiment {} is now running".format(id) + clr.ENDC
    if args.dodag:
        agg = subprocess.Popen(shlex.split("serial_aggregator -l {},{},{}".format(site, args.architecture, '+'.join(args.intermediate))), stdout=subprocess.PIPE, stderr=subprocess.STDOUT)
        threading.Thread(target=aggregator, args=(agg, pos, uid,)).start()
        time.sleep(8)
    ready(user, id)
except Exception:
    print clr.FAIL + clr.BOLD + "Stopping all experiments (Exception)" + clr.ENDC
    run_command("iotlab-experiment stop", True, True)
    print clr.FAIL + clr.BOLD
    traceback.print_exc()
    os._exit(1)

fsaintma, 2021-01-25:
Hi @souhilahana 
Thanks for your interest. You said that the script blocked with `run_command ("iotlab-experiment stop", True, True)`. I suppose that `run_command` is a function written by you. So should we have the code of this function and we can try to find a solution.
Best regards.

souhilahana, 2021-01-25:
Hi sir,
thank you for the respence, this is my script python