iot-lab/iot-lab #291: Set CONTIKI path in my project's Makefile to be able to upload my own code in M3-node

TshegoMangole, 2021-03-11:
Hi,

I have sent couple of emails previously asking for help with this matter, this is my last hope.

I want to test my own algorithm in the M3 node. So far I have setup the Contiki firmware using the tutorial ([https://www.iot-lab.info/legacy/tutorials/contiki-compilation/](url)). I am able to submit an experiment with this firmware.

However, I have my own algorithm (C++) in my own PC that I want to upload in the M3 node and test algorithm's performance. I followed this issue on **How to Upload my code in FIT/IoT-LAB**( [https://github.com/iot-lab/iot-lab/issues/220](url)) but I cannot get this "_then in your makefile, in your project, CONTIKI=path_to_iotlab_contiki"_ to work.

- Does this mean I have to copy my code into /iot-lab/parts/contiki lab using SSH to be able to set the CONTIKI path?
- How can I run the code/executable in the M3 node?

Can someone kindly assist?

Your efforts will be highly appreciated.

fsaintma, 2021-03-11:
Please follow the example of the contiki-compilation web page.
1. go to the examples directory `iot-lab/parts/contiki/examples/iotlab` 
2. copy/paste the firmware `03-sensors-collecting` (cp -R 03-sensors-collecting <your_firmware_dir>)
3. modify in <your_firmware_dir> the Makefile with good values (change CONTIKI_PROJECT/SOURCEFILES)
4. implement your algorithm in a C file
5. compile your firmware and upload binary firmware on IoT-LAB testbed.

TshegoMangole, 2021-03-17:
Thank you for your help, but I still cannot get it to work.

I followed the following steps:
1. setup iot contiki in FIT IOT-LAB SSH server following https://www.iot-lab.info/legacy/tutorials/contiki-compilation/
2. copied my own project to the FIT IOT-LAB SSH server and pasted in the _same_ directory level of the contiki firmware (which is `03-sensors-collecting`)
3. in my copied project's makefile I added the following:
`CONTIKI = ~/iot-lab/parts/contiki` 

now my project makefile looks like the following:

```
CONTIKI_PROJECT += sensors-collecting
CONTIKI_SOURCEFILES += sensors.c

output: Main.cpp
	g++ Main..cpp -O3 -D_MAIN_ -o MAIN

all: $(CONTIKI_PROJECT) output

CONTIKI = ~/iot-lab/parts/contiki

include $(CONTIKI)/Makefile.include
```

**Note: `Main.cpp` gives me an executable as an output. This is the executable i want to run inside M3-node.**
Now with the changes I added to my project makefile I cannot get firmware `sensors-collecting.iotlab-m3` which i will then upload to the IoT-LAB testbed like you mentioned in your answer. Meaning I cannot collect the pressure, and light data without this firmware.

Is there something I missed?

TshegoMangole, 2021-06-02:
Hi,
I have not been able to point my project's makefile to the contiki path. I followed the steps previously mentioned.

Is it possible to execute an output file(executable file) inside M3-node? or do I have to make the output of my algorithm a dynamic library?

Thank you.