iot-lab/iot-lab #292: steps to configure the B-L072Z-LRWAN1 board with the LoRa-net/loramac-node firmware?

shabaz-git, 2021-03-15:
Hi, 
I am using the FIT IOT-LAB for running the semtech loramac-node application. 
I am selecting the _st-lrwan1_ among the architectures and _saclay_ site. I am flashing the firmware but if I open the terminal no prints are coming. 
__[__[{{[   like these garbage prints are only visible. What might the reason?

aabadie, 2021-03-15:
Hi,

I don't the loramac-node firmware that much but I think this is a baudrate issue: on iot-lab, the stdio UART is configured at 115200 baud but the Loramac node firmware uses 921600:
https://github.com/Lora-net/LoRaMac-node/blob/f751343afd2ab2a72134438a681244b7c6f7b1cf/src/boards/B-L072Z-LRWAN1/board.c#L128

If you modify the line above with 115200 in the loramac-node code, maybe that will work.

shabaz-git, 2021-03-19:
Thanks, @aabadie . It's solved my problem. They should have given it as an argument instead of hardcoding it.

closed 2021-03-19 by shabaz-git