iot-lab/iot-lab #293: lorawan gateway(sx1301) deployment error?

shabaz-git, 2021-03-19:
Hi, 
When I am trying to select the lorawan gateway architecture, this is what happens. Even the hardware state is **_available_** and **_alive_**

[image]

What might be the reason?
Thanks.

fsaintma, 2021-03-19:
Hi @shabaz-git 
Indeed the LoRA gateway is in error state. I have just rebooted it and it should be ok now.
Can you try again please ?
Best regards.
Frederic

shabaz-git, 2021-03-22:
Hi ,
I want to see traffic from lora node st-lrwan1 to lora gateway (Sx1301). I have done steps mentioned in link "https://iot-lab.github.io/docs/boards/lora-gateway/" . I am able to see logs of nodes at terminal but not able to see logs for gateway. Could you please let me know the possible steps to configure lora gateway(S1301).

Thanks.

shabaz-git, 2021-03-22:
Hi, 
One more question. Is there any default OS running on raspberry pi 3 board(lora gateway) or we need to upload using yocto project?
If already available, how to access the lora gateway?
Thanks.

aabadie, 2021-03-22:
Hi,

First, thanks for trying the lora-gw board. This is an "experimental" feature and it's not very stable.
The Lora packet forwarder was not running on the lora-gw-1 node, that's why you couldn't see any lorawan packet in the mosquitto client.

But I just fixed the issue and now the gateway is able to catch lorawan packets again (for example, you can book another st-lrwan1 node and join a TTN network: you'll get MQTT packets from lora-gw-1 as described in https://iot-lab.github.io/docs/boards/lora-gateway).

For the moment the default OS on the lora-gw cannot be changed, it's managed by us, see the IoT-LAB Yocto image [here](https://github.com/iot-lab/iot-lab-yocto).

shabaz-git, 2021-03-22:
Hi,
Thanks for clarification.
Actually, I saw a "flash firmware" option for lora gw as well like how it is for node. So I was thinking, after building image through yocto project  I can flash through this option. And one more thing, here on gateway we are running chirpstack bridge , so that means can we see packet traffic on chirpstack as well.
Thanks.

shabaz-git, 2021-03-25:
Hi,
It seems lora gateway is still under development.
Could you please tell us possible features of lora gateway(Sx1301) which we can use for experimentation.
Thanks.

aabadie, 2021-03-27:
> And one more thing, here on gateway we are running chirpstack bridge , so that means can we see packet traffic on chirpstack as well.

Yes, but you need a specific setup:
- an SSH tunnel between the gateway and the chirpstack server: use `ssh -L 1883:lora-gw-1:1883 <login>@saclay.iot-lab.info`
- a version of the chirpstack server compatible with the gateway bridge that is running on the lora-gw device. Current version is 2.6.0 (see the [yocto recipe](https://github.com/iot-lab/iot-lab-yocto/blob/master/meta-iotlab/recipes-support/lora-gateway-bridge/lora-gateway-bridge_2.6.0.bb), so a bit old. The MQTT of the bridge has changed in more recent version and is not compatible anymore with this 2.6.0. The upgrade is on our roadmap, just need time to do it :)
- Configure the chirpstack server to interact using MQTT with the Mosquitto broker running on lora-gw-1.

> Could you please tell us possible features of lora gateway(Sx1301) which we can use for experimentation.

There's not much but you can already subscribe to the rx MQTT topic and receive all decodable lorawan packets. But you can't decode their content unless you know the AES session keys. You can also publish to the tx topic and send lorawan packets.
Some people also succeeded in connecting a chirpstack server running in the cloud to this gateway (as explained above) and could communicate with the st-lrwan1 boards deployed around (since those ones are fully programmable).

shabaz-git, 2021-04-02:
Hi,

Thanks for suggestion. we are planning to move traffic from gateway to chirpstack server as per previous discussion.we are also using same MQTT approach.
Meanwhile we tried to use
 `mosquitto_pub -h lora-gw-1 -t gateway/lora-gw-1/tx -m "downlink_msg"` 
and expected to see same message on FIT-IOT node terminal as a downlink message but we did not get it. 
Whether we can schedule downlink message from gateway to node?

Thanks.

aabadie, 2021-04-02:
> Whether we can schedule downlink message from gateway to node?

I never tried that but I think it's possible. You probably have to format/encode a valid lorawan frame (with AES encryption, etc) and use that as payload. If you look at what the gateway publish in the rx topic, I guess the tx format should be the same.

shabaz-git, 2021-04-02:
Hi @aabadie,

what we are trying to do is to communicate the lora-gw-1 with node in downlink way and for this we tried to send the message over the topic `-t gateway/lora-gw-1/tx` and the message for instance we used the same log what we are getting on the topic down below.
```
$ mosquitto_sub -h lora-gw-1 -t gateway/lora-gw-1/rx
{"rxInfo":{"mac":"b827ebfffec6b1f4","timestamp":2944906379,"frequency":868100000,"channel":0,"rfChain":1,"crcStatus":1,"codeRate":"4/5","rssi":-39,"loRaSNR":10.8,"size":20,"dataRate":{"modulation":"LORA","spreadFactor":7,"bandwidth":125},"board":0,"antenna":0},"phyPayload":"QPy3CyYAOwACzkzaob1JB/Y34SA="}
```
what we tried 
```
$ mosquitto_pub -h lora-gw-1 -t gateway/lora-gw-1/tx -m "{"rxInfo":{"mac":"b827ebfffec6b1f4","timestamp":2944906379,"frequency":868100000,"channel":0,"rfChain":1,"crcStatus":1,"codeRate":"4/5","rssi":-39,"loRaSNR":10.8,"size":20,"dataRate":{"modulation":"LORA","spreadFactor":7,"bandwidth":125},"board":0,"antenna":0},"phyPayload":"QPy3CyYAOwACzkzaob1JB/Y34SA="}"
```

We are expecting this should happen in a way that how the manually scheduled downlink from TTN works. 
What we are expecting is though payload format may wrong, atleast dio RXDONE interrupt should occur. we kept print in dio interrupt function but not getting.
we are struggling to integrate with chirpstack , do you have any steps for the same? It will be great help.
Thanks.

aabadie, 2021-04-02:
I don't know what firmware you are using on the receiving node side but before trying the full lorawan stack there, you could simply flash a firmware with a basic lora radio support in continuous listening mode (with the same frequency/bw/dr/cr parameters and check what is received). You could also try the rtl-sdr dongles deployed in Saclay to see if a radio signal is catched, that will show that the gateway is actually sending something.

shabaz-git, 2021-04-14:
> There's not much but you can already subscribe to the rx MQTT topic and receive all decodable lorawan packets. But you can't decode their content unless you know the AES session keys. You can also [...]

Till now we are able to establish the uplink (join-request) to the chirpstack network server and also able to get the accept message from the server. But, inorder to send the join-accept message to the gateway from the server we are unable to find a proper json format of the downlink message to publish on the tx topic. Can you please provide the json format compatible to **GwBridge-2.6.0** and also GwBrigde2.6.0 compatible versions of network server and application server.