iot-lab/iot-lab #294: Lorawan OTAA Join is failing

arinjay-git, 2021-03-31:
Hi, 
I am using RIOT stack specially these applications (RIOT/tests/pkg_semtech-loramac/   , RIOT/examples/lorawan/ ).
Earlier these applications were working fine but today suddenly stopped working. OTAA Joining itself getting failed though the applications are not changed. 
Is anyone faced this kind of issue? what might be the reason?

Thanks.

aabadie, 2021-03-31:
Hi,

The gateway deployed on site was down. It's now restarted and LoRaWAN join/tx should work again.
Sorry for the inconvenience.

arinjay-git, 2021-04-02:
Thank you very much.

arinjay-git, 2021-05-12:
Hi aabadie,
Could you please check once whether deployed gateway is up or not?
We are trying to use lorawan node. 

Thanks,
Arinjay

fjmolinas, 2021-05-17:
Hi, I'm experiencing similar issues where the join request is correctly received by the gateway but not on the node, I've tested on the current RIOT master as well as previous releases with the same results @aabadie @fsaintma could there be an issue with the iot-lab gateway? Cheers

fsaintma, 2021-05-17:
Hi @arinjay-git @fjmolinas 
You are right, we encounter a problem with the IoT-LAB TTN gateway. We will try to migrate the gateway with the new TTN backend server (v3). We keep you informed.

aabadie, 2021-05-24:
Hi,

I tested a bit the LoRaWAN setup in IoT-LAB and here are the conclusion:
- With TTNv2, OTAA activation indeed no longer works and unfortunately it seems there's nothing we can do to solbe this
- With TTNv2, device registered with ABP activation are working
- With TheThingsStack cloud (use the [EU1 community server](https://eu1.cloud.thethings.network/), everything works (OTAA and ABP activation modes). We recommend that you start migrating to this application backend instead of the TTNv2 backend.

aabadie, 2021-09-22:
Hi,

FYI the TTN LoRaWAN gateway on the Saclay site has been migrated and is now connected to the TTNv3 EU1 Community server and works.
Things should work as before (including MQTT and Cayenne integration) with new features: support for new versions of MAC (1.1), multicast, etc.
The notebooks on https://labs.iot-lab.info were also updated.

fjmolinas, 2021-09-22:
> Hi,
>
> FYI the TTN LoRaWAN gateway on the Saclay site has been migrated and is now connected to the TTNv3 EU1 Community server and works.
> Things should work as before (including MQTT and Cayenne [...]

Awesome thanks a lot!

ThenukaRamesh, 2021-09-26:
Hi,

I am trying to use RIOT/lorawan/ttn_getting started but OTAA connection is failing. 
Is there anyone facing the same issue?

Thanks.

aabadie, 2021-09-27:
Hi, for some reason, the TTN gateway was disconnected. Everything should be back to normal. Let us know if it works for you now.

fjmolinas, 2021-09-27:
> Hi, for some reason, the TTN gateway was disconnected. Everything should be back to normal. Let us know if it works for you now.

Confirmed it was working for me.

ThenukaRamesh, 2021-09-27:
> Hi, for some reason, the TTN gateway was disconnected. Everything should be back to normal. Let us know if it works for you now.

Thank you it is working now.

HADJMOHAND, 2024-09-26:
Hi,

I am trying to use RIOT/lorawan/ttn_getting started but OTAA connection is failing.
Is there anyone facing the same issue?

Thanks.