iot-lab/iot-lab #295: Problem : USE_NEWLIB_NANO==1 but nano include folder not found!

sabriCHALLOUF, 2021-04-02:
When I compile the gnrc_border_router for the iotlab-m3 board like usual, the compilation process doesn't start due to an error related to the newlib_nano.
This appears on the Saclay and the Grenoble sites. The same compilation passed every time without errors before one week.
When compiling for the native board, the compilation passes correctly.
RIOT used version is 2021.01
[image]

aabadie, 2021-04-02:
Building RIOT on the SSH frontend requires a recent ARM gcc toolchain, the default is 4.9 and is not compatible. You just have to issue the `source /opt/riot.source` and you will have arm gcc 9:

```
abadie@saclay:~$ arm-none-eabi-gcc --version
arm-none-eabi-gcc (GNU Tools for ARM Embedded Processors) 4.9.3 20150303 (release) [ARM/embedded-4_9-branch revision 221220]
Copyright (C) 2014 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

abadie@saclay:~$ source /opt/riot.source 
abadie@saclay:~$ arm-none-eabi-gcc --version
arm-none-eabi-gcc (GNU Arm Embedded Toolchain 9-2020-q2-update) 9.3.1 20200408 (release)
Copyright (C) 2019 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

After that the build will be successful.

Note that this is documented on [the RIOT page of the IoT-LAB website](https://www.iot-lab.info/docs/os/riot/).

sabriCHALLOUF, 2021-04-25:
Thank you
Resolved.
It is recommended to add the instruction (source /opt/riot.source) in the .bashrc file to avoid its execution every connexion.

closed 2021-04-25 by sabriCHALLOUF