iot-lab/iot-lab #296: LoRaWAN gateway migration

fjmolinas, 2021-05-17:
It seems that at the end of the year The Things Network will be dropped in favor of The Things Stack Community Edition. So it seems that gateways need to be migrated as well as devices. Are there migration plans already in place? It seems the gateway can be directly migrated or use a packet broker https://www.thethingsindustries.com/docs/reference/packet-broker/. This could allow users to slowly migrate to TTS and at the end of the year migrate the gateway. Cheers

[1] https://www.thethingsnetwork.org/docs/the-things-stack/migrate-to-v3/

fjmolinas, 2021-05-17:
@aabadie might be interested.

aabadie, 2021-05-24:
> Are there migration plans already in place? It seems the gateway can be directly migrated or use a packet broker https://www.thethingsindustries.com/docs/reference/packet-broker/. This could allow users to slowly migrate to TTS and at the end of the year migrate the gateway.

The gateway deployed in IoT-LAB can already be used with the [EU1 community server](https://eu1.cloud.thethings.network) on the TheThingsStack cloud: just create an application there and register devices and it will work. This backend server is configured by default to use the packet broker that will redirect v2 packets to the v3 stack (see [here](https://www.thethingsindustries.com/docs/getting-started/migrating/major-changes/#gateway-coverage)).

[As recommended by TTN](https://www.thethingsnetwork.org/docs/the-things-stack/migrate-to-v3/migrate-gateways/), we will migrate this gateway as late as possible, in order to allow IoT-LAB users (including us :) ) to migrate their own applications to the new stack.

> Warning:
We strongly recommend The Things Network community members to keep their gateways registered on The Things Network V2 for as long as possible, or to agree on performing coordinated migration to The Things Stack, as this will ensure the community does not loose their LoRaWAN coverage.

fjmolinas, 2021-11-05:
This one is fixed alrady.

closed 2021-11-05 by fjmolinas