iot-lab/iot-lab #299: Missing iotlab UIDs

fkerem, 2021-06-26:
Hi,

Some nodes in Saclay site has missing UIDs:

[image]

How can I access to them?
I found this [script](https://github.com/iot-lab/iot-lab/blob/master/tools_and_scripts/platform_uids_scripts/get_platform_nodes_uid.py) but couldn't make it run.

```
./get_platform_nodes_uid.py --site saclay --a8
Traceback (most recent call last):
  File "./get_platform_nodes_uid.py", line 15, in <module>
    from fabric.api import env
ImportError: No module named fabric.api
```

Thank you very much,

fkerem, 2021-06-26:
Also, found [this script](https://github.com/iot-lab/openlab/blob/master/appli/iotlab_examples/print_uids/print_uids.c) but couldn't figure out how to run it.

fsaintma, 2021-06-28:
Hi,
I have just updated the Saclay A8 uids nodes. Indeed it's not really easy to use this script based on an obsolete Fabric Python library :)
There are still missing uids with nodes that don't work (state=Dead) Currently we don't have a team member who can go on site to repair the nodes. This will wait until September.
Best regards.
Frederic

closed 2021-06-28 by fsaintma