iot-lab/iot-lab #308: Roadmap for Python 3.8/3.9

chrysn, 2022-08-22:
Upstream support for Python 3.7 expires in June 2023, which means that around that date (possibly even end of 2022), libraries might start dropping support for that version. (Concrete point: For aiocoap, which is often used in combination with RIOT experiments on encrypted CoAP communication, IoT-LAB will soon be the last common usage platform for which 3.7 is [kept supported](https://github.com/chrysn/aiocoap/issues/275)).

Do you have plans already for upgrading the SSH nodes to Debian bullseye, or would you (given that AIU it's a larger endeavour) consider starting that within the next half year?

Thanks for providing IoT-LAB!

chrysn, 2023-05-24:
The end of security support for Python 3.7 is coming up in about a month from now, and around the same time buster will become oldoldstable (with LTS support still active for a year), and Python packages start dropping 3.7 support already.

aabadie, 2023-06-15:
Hi, I think it's possible (someone did it in the past IIRC) to install miniconda on the frontend and then you'll have the version of Python you want.
[Pyenv](https://github.com/pyenv/pyenv) might also be worth a try.