iot-lab/iot-lab #310: Serial Aggregator

prathje, 2022-12-01:
Dear IoT-Lab Team,
I am trying to save the serial output of my experiments using the serial_aggregator over the SSH connection.
However, when I run it (without any arguments), I simply get the output "SerialAggregator: Empty nodes list []".
Do I really need to specify the concrete nodes? When I look into the code, it seems like it should normally query all nodes from the experiment, or am I mistaken?

Thank you for the clarification!
Best Regards,
Patrick

fsaintma, 2022-12-02:
Hi @prathje,
I have just tested an example on Lille site with success (view output below)
So I suppose that there is another problem. For example are you sure that your experiment is Running before launching serial_aggregator ? If it's the case can you give me explanations about your experiment setup, some output of commands ?

```
saintmar@lille:~/RIOT/examples/hello-world$ iotlab-experiment submit -d 120 -l lille,dwm1001,10-14,bin/dwm1001/hello-world.elf
{
    "id": 347609
}
saintmar@lille:~/RIOT/examples/hello-world$ serial_aggregator 
1669971512.004263;Aggregator started
1669971514.553347;dwm1001-12;main(): This is RIOT! (Version: 2021.04)
1669971514.553595;dwm1001-12;Hello World!
1669971514.553792;dwm1001-14;main(): This is RIOT! (Version: 2021.04)
1669971514.554484;dwm1001-10;main(): This is RIOT! (Version: 2021.04)
1669971514.554628;dwm1001-10;Hello World!
1669971514.554801;dwm1001-13;main(): This is RIOT! (Version: 2021.04)
1669971514.554937;dwm1001-13;Hello World!
1669971514.555333;dwm1001-14;Hello World!
1669971514.555661;dwm1001-11;main(): This is RIOT! (Version: 2021.04)
1669971514.555754;dwm1001-11;Hello World!
1669971514.558787;dwm1001-12;You are running RIOT on a(n) dwm1001 board.
1669971514.559111;dwm1001-12;This board features a(n) nrf52 MCU.
1669971514.559982;dwm1001-10;You are running RIOT on a(n) dwm1001 board.
1669971514.560102;dwm1001-11;You are running RIOT on a(n) dwm1001 board.
1669971514.560235;dwm1001-13;You are running RIOT on a(n) dwm1001 board.
1669971514.560383;dwm1001-10;This board features a(n) nrf52 MCU.
1669971514.560888;dwm1001-14;You are running RIOT on a(n) dwm1001 board.
1669971514.561290;dwm1001-14;This board features a(n) nrf52 MCU.
1669971514.561429;dwm1001-13;This board features a(n) nrf52 MCU.
1669971514.564163;dwm1001-11;This board features a(n) nrf52 MCU.
``

prathje, 2022-12-02:
@fsaintma Thank you very much for the prompt response. I was using the web interface so far. I will try your example once the nodes are free again.

prathje, 2022-12-02:
Sadly, the ssh approach did not help. But I am able to run the given example without any problems. I thus suspect that my firmware is not configured properly?
I am writing directly to the serial interface and I can read the serial output in the web interface without any issues. 
However, trying to listen to the node's ports did also not succeed. For an easy example, I tried the hello world example of Zephyr (2.7.2) which I also cannot get the output of using the serial_aggregator.
What are the requirements for the serial_aggregrator? Is there any other way I can easily access the serial logs of the nodes?

fsaintma, 2022-12-02:
The good way to test your firmware is to launch an experiment with it and read the serial port of one node directly from the SSH frontend with netcat command. Thus the serial port is forwarded by the IoT-LAB node with a TCP socket on port 20000.
```
> ssh <login>@lille.iot-lab.info
> nc dwm1001-<x> 20000
``` 
Be careful that the zephyr hello world firmware print only once the string at the start-up of the node. So to be sure you can reset the experiment nodes after launching the netcat command in another terminal and check that your firmware well-written on the serial port
```
> ssh <login>@lille.iot-lab.info
> iot-lab-node --reset
```

prathje, 2022-12-02:
Hi @fsaintma, thank you very much, this is what I tried with the hello world firmware.
For example, `nc dwm1001-4 20000` results in an unknown host.
I tried it with the following command, which resulted in a Connection refused error.

```
nc dwm1001-4.lille.iot-lab.info 20000
custom-35.lille.iot-lab.info [172.16.54.35] 20000 (?) : Connection refused
```

prathje, 2022-12-02:
Okay, I was not aware that there are several frontend ssh servers for each location. ;)
I was logged in at grenoble which obviously could not find the nodes but had no issues to find the experiment.
Thank you very much for your support! (a hint for this would be nice in the docs though)

closed 2022-12-02 by prathje

prathje, 2022-12-02:
I created a pull request for other "unaware" people like me ;)
https://github.com/iot-lab/iot-lab.github.io/pull/9

fsaintma, 2022-12-02:
Thanks I will check and merge it :)